softwarecrash/Solar2MQTT-ESP8266 #3: no data display

lucasimons, 2021-03-23:
I have a easun 24v 2,4kw inverter , i change this value 
// PCM and PIP inverters use below
QpiMessage _qpiMessage = {97};
QpigsMessage _qpigsMessage = {110};
QmodMessage _qmodMessage = {5};
QpiwsMessage _qpiwsMessage = {36};
QflagMessage _qflagMessage = {0};
QidMessage _qidMessage = {0};

but i see all value to 0 

the mqtt works great

wifi = { "FreeRam": 37464, "rssi": -43, "dbm": 100}
Info = {"Status":"Im alive!", "DeviceType": "PIP","IP ":"192.168.0.182"}

softwarecrash, 2021-03-24:
can you connect a second uart  (9600) to to d4 to read the debug messages?

lucasimons, 2021-03-24:
Yes I can...

I change only this value...

// PCM and PIP inverters use below
QpiMessage _qpiMessage = {97};
QpigsMessage _qpigsMessage = {110};
QmodMessage _qmodMessage = {5};
QpiwsMessage _qpiwsMessage = {36};

This value is correct because I use an other software and it works but I must use a raspberry butI  want use a wemos 

This value i don't know if are correct 
QflagMessage _qflagMessage = {15};
QidMessage _qidMessage = {18}

 I must change other variables?

I can try different value of

QflagMessage _qflagMessage = {X};
QidMessage _qidMessage = {Y};

softwarecrash, 2021-03-24:
Dont Change any of this, that are Not values.

On my Inverter ist works, so Take a Look in the debug what your Inverter ändert in the qpigs command.

lucasimons, 2021-03-24:
Ah ok so I only must set the inverter type to pip or pcm?

softwarecrash, 2021-03-24:
At the Moment only PCM is supported, i dont can Test ist on Other inverters. When you have the easun 2,4 so i think it Work. 
The best way to Check is to Look at the serial1 this Show the answer from the Inverter.

I have updated some Code, so get a fresh Copy to Test it

lucasimons, 2021-03-25:
I soldered two cables to gnd and D4 now I have to connect a usb ttl converter to my PC ... Do you have a code to test to see how the inverter responds?

softwarecrash, 2021-03-25:
> I soldered two cables to gnd and D4 now I have to connect a usb ttl converter to my PC ... Do you have a code to test to see how the inverter responds?

just load putty
connect the esp to power and the inverter and then start putty then you can see the second serial output

[image]

lucasimons, 2021-03-25:
Putty show this in loop

▒▒▒:0,"busV":0,"heatSinkDegC":0,"battV":0,"battPercent":0,"battChargeA":0,"battDischargeA":0,"sccBattV":0,"solarV":0,"solarA":0}
Ajax Request answer:
{"gridV":0,"gridHz":0,"acOutV":0,"acOutHz":0,"acOutVa":0,"acOutW":0,"acOutPercent":0,"busV":0,"heatSinkDegC":0,"battV":0,"battPercent":0,"battChargeA":0,"battDischargeA":0,"sccBattV":0,"solarV":0,"solarA":0}
Sent Command: QPIGS
Ajax Request answer:
{"gridV":0,"gridHz":0,"acOutV":0,"acOutHz":0,"acOutVa":0,"acOutW":0,"acOutPercent":0,"busV":0,"heatSinkDegC":0,"battV":0,"battPercent":0,"battChargeA":0,"battDischargeA":0,"sccBattV":0,"solarV":0,"solarA":0}

the only modification I have in the circuit are the LEDs but I don't think they create problems

softwarecrash, 2021-03-25:
the led is relict from the fork, i dont have rework the complete code at the moment.

this rs232-ttl inverter is gabage, that dint work for me, i have the one with d9 sub jack.

"Sent Command: QPIGS" after this comes the message from the inverter, so your inverter not answer to the qpigs command, or the communication  is wrong, i think the rs232 communication its not correct

lucasimons, 2021-03-25:
yes this converter i think that is garbage ..... now i buy the converter like yours

this are the answere of my inverter

Sent Command: QPIGS
Sent Command: QPIGS
Data sent to MQTT SERver - up: Sent Command: QPIGS
Sent Command: QPIGS
Sent Command: QPIGS
Sent Command: QPIGS
Sent Command: QPIGS
Sent Command: QPIGS
Data sent to MQTT SERver - up: Sent Command: QPIGS
Sent Command: QPIGS
Sent Command: QPIGS
Sent Command: QPIGS
Ajax Request answer:
{"gridV":0,"gridHz":0,"acOutV":0,"acOutHz":0,"acOutVa":0,"acOutW":0,"acOutPercent":0,"busV":0,"heatSinkDegC":0,"battV":0,"battPercent":0,"battChargeA":0,"battDischargeA":0,"sccBattV":0,"solarV":0,"solarA":0}
Ajax Request answer:
{"gridV":0,"gridHz":0,"acOutV":0,"acOutHz":0,"acOutVa":0,"acOutW":0,"acOutPercent":0,"busV":0,"heatSinkDegC":0,"battV":0,"battPercent":0,"battChargeA":0,"battDischargeA":0,"sccBattV":0,"solarV":0,"solarA":0}
Ajax Request answer:
{"gridV":0,"gridHz":0,"acOutV":0,"acOutHz":0,"acOutVa":0,"acOutW":0,"acOutPercent":0,"busV":0,"heatSinkDegC":0,"battV":0,"battPercent":0,"battChargeA":0,"battDischargeA":0,"sccBattV":0,"solarV":0,"solarA":0}

now i can change with other converter and try if it is defective

softwarecrash, 2021-03-25:
i have tried it with this little rs232-ttl pice and nothing work, with the bigger version with D9 jack it works.

i will look later, i think the little converter has another pinout from the chip.

the log show it clear, that the command is send, but no answer, any answer will show up in this.

(sorry my english is not the best)

lucasimons, 2021-03-25:
I made a mistake in connecting the wires from the rj45 to the converter now I do a further test and fix the cables because I have inverted the white one with the red one

softwarecrash, 2021-03-25:
i guess... just for test, can you connect the red and with cable to the underside of the pcb? it looks like the pads from top ar not connected to bottom, or messaure it

lucasimons, 2021-03-25:
I tried with the multimeter to see the continuity actually the two sides are not connected together, I tried to connect the cables from the other side but it doesn't work ... tomorrow the new converter should arrive I will do the test with that

softwarecrash, 2021-03-25:
ok i have checked it,
with the little converter you musst both connect on the bottom. the top are other signal lanes.
like this
https://github.com/softwarecrash/Solar2MQTT/blob/master/images/drawing.png

lucasimons, 2021-03-25:
I try to solder to the button side but there isn't any answer from the inverter 
[image]

softwarecrash, 2021-03-25:
> I try to solder to the button side but there isn't any answer from the inverter
> Uploading IMG_20210325_203354.jpg…

with your image is some wrong

lucasimons, 2021-03-25:
Yes now I re-upload the image... Your small convert works with the cables on the back?

softwarecrash, 2021-03-25:
> Yes now I re-upload the image... Your small convert works with the cables on the back?

I dont have Test it with the small

softwarecrash, 2021-03-26:
can you show me a link to your inverter or pictures?

lucasimons, 2021-03-26:
I have this
http://www.easunpower.com/Products/Off-Grid/241.html

I think that is the same of this pip 2424mse1 
https://www.mppsolar.com/v3/pip-hsemse-series/

softwarecrash, 2021-03-26:
> 
> 
> I have this
> http://www.easunpower.com/Products/Off-Grid/241.html
> 
> I think that is the same of this pip 2424mse1
> https://www.mppsolar.com/v3/pip-hsemse-series/

thanks,
ok lets go a little bit deeper.  have you the watchpower software and and can connect over serial?
the software have the option when right click on the taskbar icon click on debug, then you become a lot of tx files with answers of every command from your inverter. please zip and upload

lucasimons, 2021-03-26:
Today the big converter arrived and I made the welds tomorrow I try it ... If anything I also do the tests with whatch power ... initially I had modified your files then after I flashed your bin file directly
[image]

softwarecrash, 2021-03-26:
> Today the big converter arrived and I made the welds tomorrow I try it ... If anything I also do the tests with whatch power ... initially I had modified your files then after I flashed your bin file directly
> [image]

You have a Wireing issued, the yellow cable is your vcc so ist Looks, but going to tx

lucasimons, 2021-03-26:
Yes I send a wrong image i connect vcc to 3.3v gnd to gnd e rx to TX e tx to rx... Now I think it is correct

lucasimons, 2021-03-27:
I connect the pc with the USB cable but I don't find any debug features on watchpower.... Where it is?

softwarecrash, 2021-03-27:
In the taskbar right click on the watchpower icon

lucasimons, 2021-03-27:
I think that I have a different version of watchpower 
[image]

softwarecrash, 2021-03-27:
[image]
the watchpower icon on the taskbar, right click and debug

lucasimons, 2021-03-28:
[debug-watchpower.zip](https://github.com/softwarecrash/Solar2MQTT/files/6216967/debug-watchpower.zip)
This is my file in the log folder

softwarecrash, 2021-03-28:
the top is yours, the bottom from me
[image]
so it looks okay, i think the problem must be on another place.
have you connect it on the pc over usb or serial? when serial the port works
just for fun, try swap the rx/tx cables from the inverter, i have seen shematics with other pinouts.

you can test the communication another way when you connmect the esp with converter to the pc com port, with putty you can see then the request from the esp, then copy one line from the qpigs file begin at ( in the console to simluate the inverter

lucasimons, 2021-03-28:
I have always used the usb port ... if I did the test with the console and putty I would only see if the esp works? what parameters should I put on putty? (data bit, stop bit, etc) ...
I had inserted the small switch to reprogram the esp via usb ...

softwarecrash, 2021-03-29:
yes, the inverter rate is all 2400 8N1

i dont have build actual an automatic function, its not easy for me to implement a automatic function, but thanks i will take al look later. my version its early alpha, so many bugs and rare feauteres.

so i hope i can work in future on it, or/and find people there help me with this

closed 2021-05-05 by softwarecrash