softwarecrash/Solar2MQTT-ESP8266 #4: PCM - Data in wrong order

ian-tait, 2021-03-28:
Hi Rocky,

I'm here because of your message on "daromer2".
You were right this fork has given me some results but they seem to be in the wrong order.

I have set the type to PCM
[image]

Is it a case of just moving some of the variables around or have I messed something up?

ian-tait, 2021-03-28:
[image]

softwarecrash, 2021-03-29:
hey, i working actual an the pip / pcm. yes just sort the variables in the inverter.cpp
but bevor, take the debug files from watchpower, there you can see of what for command your inverter answer and the order of data, please share this, so i can eventualy make a more dynamic thing

closed 2021-05-05 by softwarecrash