softwarecrash/Solar2MQTT-ESP8266 #6: only QPIGS

mihaigsm2003, 2021-08-31:
i test software on MPP Solar GK model, and i see only QPIGS command request. other command not sent.

softwarecrash, 2021-10-06:
When the qpigs command Not answered succesfull Other commands dont send

mihaigsm2003, 2021-10-20:
> 
> 
> > good, i have version with all this data, not for public, i can see you if u need [http://www.gsx.ro/dashboards](url)
> 
> nice, but i have build this project with target to get the data in my smarthome ovber mqtt, i only need it to manage the power regualtions of the house. so im not a pro programmer, so this in the acutal state cost me a lot of time

i am not a pro programmer too, i spent lot of time too. any way good job!

softwarecrash, 2021-10-20:
> i am not a pro programmer too, i spent lot of time too. any way good job!

when you have time and fun, you can help make this little thing better, i have a lot ideas but not a lot knowledge

softwarecrash, 2021-10-20:
> 
> 
> > > > > good, i have version with all this data, not for public, i can see you if u need [http://www.gsx.ro/dashboards](url)
> > > > 
> > > > 
> > > > nice, but i have build this project with target to get the data in my smarthome ovber mqtt, i only need it to manage the power regualtions of the house. so im not a pro programmer, so this in the acutal state cost me a lot of time
> > > 
> > > 
> > > i am not a pro programmer too, i spent lot of time too. any way good job!
> > 
> > 
> > when you have time and fun, you can help make this little thing better, i have a lot ideas but not a lot knowledge
> 
> sure, but i use diferent mode:
> 
> ` if (Serial.available() <= 0) { Serial.print(QVFW); // send an initial string RESQUEST delay(100);}else {} stringZr = Serial.readString(); //Serial.println (stringOne);
> 
> if (Serial.available() <= 0) { Serial.print(QPIGS); // send an initial string RESQUEST delay(100);}else {} stringOne = Serial.readString(); //Serial.println (stringOne);
> 
> if (Serial.available() <= 0) { Serial.print(QMOD); // send an initial string RESQUEST delay(100);}else {} stringTwo = Serial.readStringUntil('\r'); //Serial.println (stringTwo);
> 
> ```
> if (Serial.available() <= 0) {
> Serial.print(QID);    // send an initial string RESQUEST
> delay(100);}else {}
> ```
> 
> stringThree = Serial.readString(); //Serial.println (stringThree);
> 
> ```
> if (Serial.available() <= 0) {
> Serial.print(QPI);    // send an initial string RESQUEST
> delay(100);}else {}
> ```
> 
> stringQPI = Serial.readString(); stringQPI.remove(0, 1); stringQPI.remove(4, 4); //Serial.println (stringQPI);
> 
> ```
>  if (Serial.available() <= 0) {
> Serial.print(QLT);    // send an initial string RESQUEST // modificat aici
> delay(100);}else {}
> ```
> 
> stringFour = Serial.readString();
> 
> ```
>   if (Serial.available() <= 0) {
> Serial.print(QET);    // send an initial string RESQUEST
> delay(100);}else {}
> ```
> 
> stringFive = Serial.readString();
> 
> ```
> stringFour.remove(0, 1);
> stringFour.remove(8, 4);
> stringFive.remove(0, 1);
> stringFive.remove(8, 4);
> //Serial.println(stringFour.length());
> //Serial.println(stringFive.length());
>   if (stringFour.length() == 8 && stringFive.length() == 8) {
> ```
> 
> //Serial.println (stringFour); //Serial.println (stringFive); } else { stringFour = "-"; stringFive = "-"; } if (Serial.available() <= 0) { Serial.print(QPIGS2); // send an initial string RESQUEST delay(100);}else {} stringSix = Serial.readString(); stringSix.remove(0, 1); stringSix.remove(16, 4); if (stringSix.length() == 16) {
> 
> } else { stringSix = "PV2 NO_DATA"; }
> 
> stringTwo.remove(0, 1); stringTwo.remove(1, 4); stringThree.remove(0, 1); stringThree.remove(14, 4); stringOne.remove(107, 4); stringOne.remove(0, 1); stringOne.replace(':', ','); stringOne.replace(' ', ','); stringOne.remove(35, 4); stringSix.replace(' ', ','); `

sure, other inverters send other command, i have seen that from an 5kw inverter (big one of mine) the ordering from pqigs is diferent, so my initial idea was to get the version number and communication protokol number from the inverter and then select the command and data ordering

mihaigsm2003, 2021-10-20:
QED, QLT work on invertor with HPV with panel detachable and all hybrid. for daily , month need to sent data with hour and date. actually not work for me, i can't set time on my esp

mihaigsm2003, 2021-10-20:
for year and monthly  u can preset on esp

closed 2021-11-09 by softwarecrash

mihaigsm2003, 2021-11-10:
i test both version but for me not work, Only inverter mode will see "Battery" also all data is 0 (zero)

[image]

i try both type, onlu inverter mode

mihaigsm2003, 2021-11-10:
[image]
only this data!

softwarecrash, 2021-11-10:
there is no serial monitor when the esp is setted up and running.
you can grab the raw data from the inverter over mqtt, thats what i need

mihaigsm2003, 2021-11-10:
No data data sent via mqtt, tested.

softwarecrash, 2021-11-10:
> 
> No data data sent via mqtt, tested.
> […](#)

there a raw folder in the mqtt topic, this field are empty? then you dont get data from the inverter or communication wrong

mihaigsm2003, 2021-11-10:
I put monitor because software is stuck in this position. Request sent one time, after stuck.

I have monitor to mqtt too, no data sent because software stopped. I set to 10s, 30s and 60 seconds only one time is sent request

softwarecrash, 2021-11-10:
> 
> I put monitor because software is stuck in this position. Request sent one time, after stuck. I have monitor to mqtt too, no data sent because software stopped. I set to 10s, 30s and 60 seconds only one time is sent request
> […](#)

what request is one send?
can you grab the logs from powerwatch software?

mihaigsm2003, 2021-11-17:
[image]
script is stop working after send this data becouse no date received. no data sent via mqtt, In your script if no data receiving script stop working.

softwarecrash, 2021-11-17:
> [image] script is stop working after send this data becouse no date received. no data sent via mqtt, In your script if no data receiving script stop working.

after finnishing the wifi setup no more data comes over the com port, the serial0 is used for communication with the inverter after wifi settings. so what you see is the raw data was send to or from the inverter, this looks like your inverter answer wrong, or the rs232 converter dosnt do her job right, or cable wireing is wrong.
the micro pice of rs232 in the description is from the other fork, i dont have tested it. but i think it should work too.

have you tried it after the wifi setup is finish to go to the web interface and set the inverter under setting to pip again and check the other data?

and again, can you provide me the watchpower pc software logfiles so that i can see what for a version and type of communication your device using?