softwarecrash/Solar2MQTT-ESP8266 #7: Receive JSON data

horatiuanghel, 2021-09-10:
Hi,

I am  testing your first release and the only JSON data that I get from the inverter is pBattV and battPercent. 
I am doing something wrong or it's a bug ? I am trying to insert the rest of the data to influxdb but it seems that I need the data as JSON (only pBattV and battPercent are in my database). Attached you can find a screenshot from MQTT explorer. 
[https://ibb.co/GCrLGgp](url)

Thank you!

softwarecrash, 2021-10-03:
my version actualy dont support full json data, only mqtt points

horatiuanghel, 2021-10-05:
ok
I saw  that the script should have also inverter operation mode. 
I am not receiving this value, even in mqtt points

horatiuanghel, 2021-10-05:
Tried with the inverters bellow and not receiving temp and inverter status
MppSolar PIP5048MGX
EAsun SML II
EAsun VE

softwarecrash, 2021-10-06:
I have only one SMV from easun, so i cant Test Other communication.
But when you can Provide the Log Files from the PC Software (right click in the taskbar Icon and select enable Log) then i can Take a Look why it Not Work, and when my time it alow make Changes to Support your devices