softwarecrash/Solar2MQTT-ESP8266 #11: Easun IGRID SV II 5 KW - All values are Zero

BMBIT-oss, 2022-07-16:
Hi , 
I have an Easun IGRID SV II 5 KW , have made all the hardware connection and it looks ok , tested in couple of ways , but all values are Zero .
Debugging is ON and I get the following :

Sent Command: QPIGS
   Calc: E300   Rx: E300
   Recived: ^0 
message recived
Sent Command: QMOD
   Calc: E300   Rx: E300
   Recived: ^0 
Sent Command: QPIGS
   Calc: E300   Rx: E300
   Recived: ^0 
Sent Command: QMOD
   Calc: E300   Rx: E300
   Recived: ^0 
Sent Command: QPIGS
   Calc: E300   Rx: E300
   Recived: ^0 
Sent Command: QMOD
   Calc: E300   Rx: E300
   Recived: ^0 
Sent Command: QPIGS
   Calc: E300   Rx: E300
   Recived: ^0 
Data sent to MQTT Server
Sent Command: QMOD
   Calc: E300   Rx: E300
   Recived: ^0 
Sent Command: QPIGS
   Calc: E300   Rx: E300
   Recived: ^0 
message recived
Sent Command: QMOD
   Calc: E300   Rx: E300
   Recived: ^0 
Sent Command: QPIGS
   Calc: E300   Rx: E300
   Recived: ^0 
message recived
Sent Command: QMOD
   Calc: E300   Rx: E300
   Recived: ^0 
Sent Command: QPIGS
   Calc: E300   Rx: E300
   Recived: ^0 
Sent Command: QMOD
   Calc: E300   Rx: E300
   Recived: ^0 

Can you point me in the wright direction please !

BMBIT-oss, 2022-07-17:
I have changed the values , like another colleague has done it hire an now I only get : grid Voltage : 110 V ( My grid Voltage is 230 V actually ):
// PCM and PIP inverters use below
QpiMessage _qpiMessage = {97};
QpigsMessage _qpigsMessage = {110};
QmodMessage _qmodMessage = {5};
QpiwsMessage _qpiwsMessage = {36}; 

So I believe is something to do with the CRC , maybe ?

BMBIT-oss, 2022-07-19:
After some digging and a bit of wiring it looks like the protocol is a bit different for EASUN IGRID SV Series .
I have done some tapping and picked up some data from the PC to Inverter :

^P005GSX
^P006FWS?C
^P006GMN?W
^P006GMN?W
^P006GMN?W
^P006MOD??
^P006VFW??
^P007PIRI?8
^P013ED20220707??
^P013ED20220708O
^P013ED20220709_8
^P013ED20220710? 
^P013ED20220711?
^P013ED20220712?b

Also I have managed to find the entire protocol description Online .

softwarecrash, 2022-07-22:
sorry, i dont get a mail that a message is here....

yes looks like your inverter have different protocol, so its possibil to make my thing to work, but its hard way. 
take a look at the older releases, that have snipets for this protocol

BMBIT-oss, 2022-07-29:
Thank you , 
It looks like this is the new protocol that Easun and equivalent will use in the future .
I have managed to modify your code and implement the P18 Protocol in to it , but is still under development and testing .

softwarecrash, 2022-07-29:
> Thank you , It looks like this is the new protocol that Easun and equivalent will use in the future . I have managed to modify your code and implement the P18 Protocol in to it , but is still under development and testing .

Sounds good.
Is the request command for the protocol Version the same?
I had the Idea vor Months to Automate select the protocol Version and commands, but i dont can Test anything with This.
When your Version is working, you ar willing to make a Switch for the Versions and implement it?

softwarecrash, 2022-08-12:
@BMBIT-oss can you provide your command table for the inverter?
or will you help modify the project to support mutliple commands and inverters?

BMBIT-oss, 2022-08-13:
I'm Not really a guru in programing in Arduino .
I have Managed to get couple of functions working and have uploaded the code on my GitHub page as per below :
https://github.com/BMBIT-oss/Solar2MQTT_P18

BMBIT-oss, 2022-08-13:
This is the protocol , it's using the same CRC an so on .
The return data is a bit different but your code its able to pull all the information with an easy modification .
[EASUN Protocol Real 18.pdf](https://github.com/softwarecrash/Solar2MQTT/files/9331524/EASUN.Protocol.Real.18.pdf)
.

softwarecrash, 2022-08-13:
Nice,
When Interesst you can Join the discord Server.
I Plan to make a new one of this programm with serveral configs for Inverter and need Help to write it

softwarecrash, 2023-09-16:
support added  with  V1.1.0

closed 2023-09-16 by softwarecrash