softwarecrash/Solar2MQTT-ESP8266 #13: IGridSV-IV5.6KW

Cryptimo, 2022-08-30:
Hier ein Logfile vom IGridSV-IV5.6KW. Bei 10:06 Uhr habe ich die Einstellung für die LED Anzeige geändert (Ist zwar unwichtig, aber um überhaupt mal eine Einstellung zu ändern). Im beigefügten LOG nach dem Stichwort "LED" suchen.

Es gibt nur noch eine Log-Datei. Früher muss es bei ähnlichen Modellen mehrere Dateien gegeben haben. Hier entsteht aber nur eine Datei.
[2022-08-30 IGridSV-IV5.6KW.log](https://github.com/softwarecrash/Solar2MQTT/files/9451213/2022-08-30.IGridSV-IV5.6KW.log)
[image]

Cryptimo, 2022-08-30:
Hier die Anleitung:

[IGridSV-IV5.6KW - Manual Ver.1.0.pdf](https://github.com/softwarecrash/Solar2MQTT/files/9451574/IGridSV-IV5.6KW.-.Manual.Ver.1.0.pdf)

softwarecrash, 2023-04-14:
komma separierte werte, kein PI30 protokol, wird später hinzugefügt

mihaigsm2003, 2023-07-26:
this is P18 protocol

rolsch, 2023-07-29:
Hi, i have an IGridSV-IV5.6KW.
How can i support this project?
Actual, i'll build the ESP-Adapter for this :-)

Ist diese Aussage noch aktuell bei den fertigen Modulen im Shop?

Die aktuell bei Github releaste Firmware funktioniert nicht mit dieser Platine!
Auch nicht der PI18-Fork, dieser wird nicht von uns gepflegt und betreut.

softwarecrash, 2023-07-29:
> Hi, i have an IGridSV-IV5.6KW. How can i support this project? Actual, i'll build the ESP-Adapter for this :-)
> 
> Ist diese Aussage noch aktuell bei den fertigen Modulen im Shop?
> 
> Die aktuell bei Github releaste Firmware funktioniert nicht mit dieser Platine! Auch nicht der P18-Fork, dieser wird nicht von uns gepflegt und betreut.

welche aussage im shop?
mit der platine funktioniert aktuell nur eine angepasste firmware, die releases gehen noch nicht.

der p18 fork geht genausowenig da dort auch eine anpassung fehlt.

funktioniert der inverter mit watchpower?

mihaigsm2003, 2023-07-29:
> Hi, i have an IGridSV-IV5.6KW. How can i support this project? Actual, i'll build the ESP-Adapter for this :-)
>
> Ist diese Aussage noch aktuell bei den fertigen Modulen im Shop?
>
> Die aktuell bei [...]

use this link for your inverter, need compile, but firs add your data in settings file. For moment i remove custom fields, but you can modify if need

softwarecrash, 2023-07-29:
> use this link for your inverter, need compile, but firs add your data in settings file. For moment i remove custom fields, but you can modify if need: [...]

this will not work with the PCB edition!

rolsch, 2023-07-29:
Hi Rocky.

 

Thanks for the info. I hope that in the near future the PCB edition works with the public github release AND with the P18 code.

 

https://github.com/mihaigsm2003/Read-Data-Inverter-P18

mihaigsm2003, 2023-07-31:
> > > Hi, i have an IGridSV-IV5.6KW. How can i support this project? Actual, i'll build the ESP-Adapter for this :-)
> > > Ist diese Aussage noch aktuell bei den fertigen Modulen im Shop?
> > > Die aktuell bei Github releaste Firmware funktioniert nicht mit dieser Platine! Auch nicht der P18-Fork, dieser wird nicht von uns gepflegt und betreut.
> > 
> > 
> > use this link for your inverter, need compile, but firs add your data in settings file. For moment i remove custom fields, but you can modify if need:
> > https://github.com/mihaigsm2003/Read-Data-Inverter-P18
> 
> this will not work with the PCB edition!
> 
> actual i rewrite the whole code for a new edition with more support for devices.... need time

why not work? i think both use same hardware. yes, my software is optimized for original interface from china , work with panel leds.

softwarecrash, 2023-07-31:
> why not work? i think both use same hardware. yes, my software is optimized for original interface from china , work with panel leds.

ah, missunderstanding.
your software, and the actual release here, will not work with the PCB that we sell. the PCB need a modified variant.
i work on it, so the code have heavy modifications. when it will work was i try, i can adapt the p18, and sometimes the other protocols too

softwarecrash, 2023-09-16:
support added  with  V1.1.0

closed 2023-09-16 by softwarecrash