softwarecrash/Solar2MQTT-ESP8266 #23: Set command dont work

den-localnet, 2023-01-29:
Hello!
http://192.168.0.XXX/set?maxcharge=20 or http://192.168.0.XXX/set?maxaccharge=20
answer "message received" but charging current don`t change 
EaSun 5.5 kva
[image]

softwarecrash, 2023-01-29:
i think your device dont understand the commands behind the setcharge, take a look at your watchpower debug output when you set the charge and then use this command with the set_command over mqtt

den-localnet, 2023-01-29:
Thank you 
Tell me how to correctly send a command through mqtt. What's the syntax?

Do you plan to debug in web interface?
Sending custom command through the web interface?
(like tasmota)

closed 2023-01-29 by den-localnet

softwarecrash, 2023-01-29:
as example my inverter need for charge the command MUCHGC020 for limiting by 20A so i can set this sting by set_command
i think your inverter have different command for.

later i will rework the solar thing a little bit