softwarecrash/Solar2MQTT-ESP8266 #24: MasterPower UM5KV4 5,6 KW / Axpert VM IV 5,6 KW

ruppik, 2023-02-03:
Hallo Rocky,
ich habe für meinen Offgrid Inverter mit Deinem Programm und der Hardware
 Wemos D1 Mini + Max3232 
einen Webmonitor gebastelt, es werden allerdings keine Daten angezeigt.
Jetzt meine Frage:
Ist dieser Inverter überhaupt vom Datenprotokoll her kompatibel mit Deinem Programm,
oder habe ich noch irgendein anderes Problem, evtl. beim RS232-TTL Converter?
Rx und TX habe ich natürlich schon hin- und hergetauscht.
Wie könnte ich testen, ob überhaupt Daten rüberkommen?
Danke!

ruppik, 2023-02-04:
Das Problem hat sich geklärt, es läuft alles wunderbar mit meinem Master Power Wechselrichter!
Woran lag's?
Am Ausgang von meinem Max3232 Modul durften TTL-seitig Tx und Rx zum D1 Mini nicht gekreuzt werden, sondern 1:1 durchverbunden. Nachdem ich das als letzten Versuch noch getestet habe, läuft es nun wunderbar.
Danke für das super Programm!!

closed 2023-02-04 by ruppik

Wox1, 2023-03-25:
> Das Problem hat sich geklärt, es läuft alles wunderbar mit meinem Master Power Wechselrichter! Woran lag's? Am Ausgang von meinem Max3232 Modul durften TTL-seitig Tx und Rx zum D1 Mini nicht [...]

Hallo, kannst Du auch Befehle senden oder nur empfangen?