softwarecrash/Solar2MQTT-ESP8266 #26: Kodak/Axpert/PIP OG5.48

dazzlersa, 2023-02-17:
Your repo looks perfect for what I am doing to link to Home Assistant, amazing work!!

What are the chances that the D1-Mini is powered and communicates on the built-in USB port D1 Mini side and then the USB output (Micro-USB) on the Inverter?

The Micro USB port works as I have had ned-Kellys' Voltronic working perfectly on it on a Pi. But it popped, unfortunately.

I am hoping the pics help to explain. Basically USB to USB.

[image]
[image]

TIA 

Daz

dazzlersa, 2023-02-18:
After reading every thread in the issues I tried the big max3232 and also tried the Rx- Tx and Rx-RX combos, still get nothing on the webpage. Any ideas?
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

dazzlersa, 2023-02-18:
Sorry they aren't in order.

softwarecrash, 2023-03-19:
sorry the late answer.
a connection usb-usb is not possibil, that idea need a usb host shielt and this is verry complex and expensive.
when xou not get data fromn the inverter please connect it over serial to the pc and start watchpower, then activate debug in watchpower, that will create a bunch of files. this files i need to check it will work

softwarecrash, 2023-04-14:
@dazzlersa work this inverter with serial in korrekt order?

softwarecrash, 2023-04-23:
no issue

closed 2023-04-23 by softwarecrash