softwarecrash/Solar2MQTT-ESP8266 #27: Masterpower Omega-UM3KVA

michael-wagner73, 2023-04-10:
Die Daten des Masterpower Omega UM3KVA werden in der Solar2MQTT V0.5.1 nicht korrekt angezeigt. Die Temperatur kommt vom Masterpower schon in den Rohdaten falsch. 

**Screenshots**

[image]
[image]
[debug.zip](https://github.com/softwarecrash/Solar2MQTT/files/11189323/debug.zip)

michael-wagner73, 2023-04-10:
Zwischen den beiden Screenshots liegen ein paar minuten Zeit, daher die unterschiede.

softwarecrash, 2023-04-14:
kannst du deinen verkäufer kontaktieren das er dir das pdf für das rs232 protokol schickt?

michael-wagner73, 2023-04-14:
Hi, mache ich. Hoffe sie schicken was.Lgm --Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

softwarecrash, 2023-04-14:
> 
>  
>   
>   
>  
>  Hi, mache ich. Hoffe sie schicken was.Lgm --Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.Am 14.04.23, 19:23 schrieb Rocky ***@***.***>:
> 
>    
>   kannst du deinen verkäufer kontaktieren das er dir das pdf für das rs232 protokol schickt? 

Super, danke.
Ich kann den Wert der als Temperatur angezeigt wird absulut nicht zuordnen.
Man könnte es jetzt so hinfriemeln das es passt, aber mit Protokoll wäre das deutlich einfachet

michael-wagner73, 2023-04-19:
Hallo zusammen,
Mir scheint unter den aktuellen Bedingungen ein Milli -Fahrenheit Wert realistisch. Wobei das einem Blick in die Glaskugel gleich kommt

michael-wagner73, 2023-04-19:
> Hallo Michael, ich habe den gleichen Hybriden. Spannend, bei mir wird die Temperatur problemlos angezeigt. Dafür kein SCC. Den gibt meiner aber tatsächlich nicht raus. Den einzigen Unterschied, den ich zwischen Deinem und meinem sehe, sind die CPU-Software-Versionen. Ich habe als Haupt die 47.00 und als Zweit die 103.24. Oder ist evtl. in Deinem der Temp-Sensor "Schrott"? Du schreibst ja, dass schon die Rohdaten vom Masterpower falsch sind.

Das der Sensor Schrott ist könnte auch sein.

michael-wagner73, 2023-04-19:
Zweite Urgenz beim Hersteller. Leider noch keine Reaktion.

softwarecrash, 2023-09-16:
support added  with  V1.1.0

closed 2023-09-16 by softwarecrash