softwarecrash/Solar2MQTT-ESP8266 #38: [FEATURE] Different Interval for important values like Solar- Watt Volt Ampere

AndyMuc72, 2023-07-03:
### Is your feature request related to a problem? Please describe

Add another interval for polling only some few values with a higher interval, that can be chosen, if the load is too high for pulling all values at the same time. Over seriell connection I pull it every 0,5 secs.
e.g.
Solar-Watt
Solar-Volt
Solar-Ampere

### Describe the solution you'd like

Solar-Watt
Solar-Volt
Solar-Ampere

### Describe alternatives you've considered

there is only the old or new way :-)

softwarecrash, 2023-07-03:
There is No way to get only this values in this frequency

closed 2023-07-03 by softwarecrash

AndyMuc72, 2023-07-04:
thank you for your great work. I love your project.
I understand that you get all the data at the same time with QPIGS and QPIRI.
I showed Nuno, the author if solpiplog, your project. He did not know it yet.

softwarecrash, 2023-07-04:
In the actual Release yes all commands requestet.
I Work on a new one that only use qpigs and qmod for variable Data and only reask the Stativ Data when Changes send.
But this Work need a Lot of time before ready