softwarecrash/Solar2MQTT-ESP8266 #39: [Bug]: Not reading W consumption values

Konycz, 2023-07-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Correct loading of inverter consumption values

### Screenshots / Fotos

From the beginning it worked for approx. 2 days, after that the consumption is loaded once in approx. 2 hours, charging the battery, etc. everything is fine
[image]

### Version

0.5.1

### Relevant livejson output

```json
Solar In:
0V 0A 0W
Grid In:
1V 0Hz
AC Out:
228V 50Hz
AC Load:
0Va 0W 0%
Bus:
391V
Temperature:
342°C
Battery:
27.19V 100% 100%
SCC | IN | Out:
0V 0A 1A
Inverter Mode:
Battery
```

### What browsers are you seeing the problem on?

Firefox

softwarecrash, 2023-07-19:
looks like your inverter use another sorting of data, your tempm value is the indicator

softwarecrash, 2023-07-19:
your inverter send the data in the wrong order, you can actual nothing do.
i work on a complete new version but that need a lot of time

Konycz, 2023-07-19:
But according to the display of the inverter, the consumption is 89-92W and this is not shown anywhere. And if by chance it shows the correct value there, at that moment the load of the converter in percentage is also shown (which also shows correctly), but only for a moment and then again on both the value is 0

And when will you release it approximately with the completion time of the new version?

softwarecrash, 2023-07-19:
sure, thats missreading of the inverter data, what for model you have?

Konycz, 2023-07-19:
https://vi.aliexpress.com/item/1005003340956819.html?spm=a2g0o.order_list.order_list_main.46.6fff18020dsd74&gatewayAdapt=glo2vnm

Konycz, 2023-07-19:
OK

Konycz, 2023-07-19:
and where is the debug mode set in that program?

closed 2023-09-16 by github-actions[bot]