softwarecrash/Solar2MQTT-ESP8266 #46: [Bug]:  does not hand in data,but it mqtt accession signals it

zsolti80, 2023-09-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Other (please write your hardware in 'What happened?')

### What happened?

Onto ready one was tackling I use a supply:
https://all-solutions.store/de/shop/solar2mqtt-set/?fbclid=IwAR3cGeClP7fJlM98zeJZKsQ3pgoEdsneDGYULCUFgtfJaek6UM9SHP44Jzw

Does not work well, it mqtt contact good, but does not show data on the web application surface. Once good, restarted later not good.
The apparatus signals it how was attached, but does not send data. But is when works and sends it. But I do not see data again now!
My inverter: Qoltec 53886 Watchpower application and wifi added to him an apparatus works.

### Screenshots / Fotos

[image]

### Version

1.x.x and above

### Relevant livejson output

```json
{"EspData":{"Device_name":"Solar2MQTT","ESP_VCC":2.981,"Version":"1.0.0"},"DeviceData":{},"LiveData":{}}
```

### What browsers are you seeing the problem on?

Chrome

zsolti80, 2023-09-14:
He works again!
But the data appear as an entity without an identifier.
How can I take these under the energy ear? How I shall see it the battery and inverter consumption?

zsolti80, 2023-09-14:
I entered the data points in the configuration.yaml file.  I have attached a picture.  But these are entities without an identifier.  Because of this, I cannot enter the data under the energy tab.  I would like help with this.  How can I display the data as an entity?  What should I put in the configuration.yaml file?
[image]

Jarnsen, 2023-09-14:
Hi,

here is an example for the configuration. Its for daly2mqtt, but the form is the same.

Watch at code2, After the you must use a Rieman helper. The Energy Dashboard accepts only energy entities no power entities.

https://github.com/Jarnsen/Config_Daly2MQTT

closed 2023-10-30 by github-actions[bot]