softwarecrash/Solar2MQTT-ESP8266 #56: [HELP] commands via MQTT not accepted

csuther28, 2023-10-27:
### A clear and concise description of what the problem is.

My inverter with Solar2MQTT works extremely well, using latest version 1.1.0 on a D1 mini.
Protocol to inverter using PI30.
Receiving all good info onto web page and also passing through via MQTT to my Openhab system.
I can also send successfully inverter commands via the web serial page successfully.
However, the inverter seems to not respond to any commands sent via MQTT using : topic/Device_Control/Set_Command
e.g. inverter2/Device_Control/Set_Command/POP00
I loaded the debug version onto D1 mini and via the web serial interface could I then see the comms between D1 mini and inverter, sending commands from the web is all good and one can see how the inverter reacts, i.e. accepting or rejecting etc.
But sending commands from MQTT, nothing comes up on the serial interface.
Does the firmware accept external commands via http? i.e. could I send commands directly to the D1 mini via http to accept commands?

derLoosi, 2023-10-27:
Hey,

please try the correct mqtt path:

 inverter2/DeviceControl/Set_Command/POP00

(without the _ in DeviceControl)

Greetings
Daniel

csuther28, 2023-10-27:
Hi Daniel.
Thanks for quick response.....ok, tried as suggested...and now works!!!
As mentioned, I used the pre-compiled 1.1.0 version, but I also looked at the source files and from their (same as on wiki) did it show the format to "Device_Control/Set_Command".
Clint.

softwarecrash, 2023-10-27:
> Hi Daniel. Thanks for quick response.....ok, tried as suggested...and now works!!! As mentioned, I used the pre-compiled 1.1.0 version, but I also looked at the source files and from their (same as on wiki) did it show the format to "Device_Control/Set_Command". Clint.

nope, tzhe code are correct
https://github.com/softwarecrash/Solar2MQTT/blob/b52f7f333a6462570633fc94facad36d06dc76a3/src/main.cpp#L547

closed 2023-12-11 by github-actions[bot]