softwarecrash/Solar2MQTT-ESP8266 #59: Protocol 17 Voltronic Mppsolar FSP needed

Steef-the-Beef, 2023-11-05:
### Is your feature request related to a problem? Please describe

Hi and thank you for your great work!

How can I implement the protocol P17 which is linked as alternative protocol?

(if you want we can speak German)

Thank you! :-)
Stefano 

### Describe the solution you'd like

implement the protocol P17

### Describe alternatives you've considered

implement the protocol P17

softwarecrash, 2023-11-06:
Hallo,
das P17 sollte im grundsatz funktionieren, zumindest ist die protokoll art eingebaut.
es kann sein das es nicht erkannt werden kann da sich da schon wieder kleinigkeiten geändert haben.
ohne ein gerät was dieses protokoll spricht da zu haben ist das immer recht schwierig.
welche P17 version hast du den?

Steef-the-Beef, 2023-11-06:
Moin,

Ich nutze einen Mppsolar MPI 15k WP.
Der nutzt das Protokoll 17. Das Protokoll 17 hat minimal geänderte Befehle zum Protokoll 18.

https://github.com/jblance/mpp-solar/blob/master/docs/protocols/PI17%20-%20Infini-Solar%20WP%20LV6K_3P%2010-15K%20and%20TWIN%20Series%20protocol%2020220921.pdf 

Genau das Protokoll nutzt mein Wechselrichter. 
Wie bekomme ich das implementiert? Nutzt du VisualStudio Code? 

Viele Grüße
Stefano

softwarecrash, 2023-11-06:
ja nutze vsc, um das p17 einzubinden müsste man ein weiteren protokol zweig anlegen sammt allem drum und dran

closed 2023-12-22 by github-actions[bot]

Steef-the-Beef, 2024-03-01:
[PI_Serial.zip](https://github.com/softwarecrash/Solar2MQTT/files/14463659/PI_Serial.zip)
Moin moin,

ich hab versucht das Protokoll 17 zu implementieren (PI_Serial). Leider passt da irgendwas nicht. Vielleicht hab ich eine Kleinigkeit übersehen.

Könnt ihr bei Gelegenheit drüber schauen?

Viele Grüße
Stefano

softwarecrash, 2024-03-01:
grüße,
so wird das nichts, mach bitte ein fork und arbeite da deine änderungen ein, so suche ich mir einen wolf was du geändert haben könntest.

"irgendwas passt nicht" ist leider auch absulut meta, da kann ich garnichts mit anfangen

Steef-the-Beef, 2024-03-01:
Moin,
Ich bin noch nicht so firm mit github, werde mir das aber anschauen.

Ich hab lediglich im Bereich PI_Serial das Protokoll PI17 eingefügt. Ich hab mich am Protokoll PI18 orientiert und habe alle Punkte so "übersetzt" wie im PI18.

Die Erkennung des Protokolls scheint aber nicht zu funktionieren, da keine Daten vom Wechselrichter ausgelesen werden. Auch auf der Webgui des esp8266 werden keine Daten angezeigt.

Muss ich außer PI_Serial noch etwas beachten, um das Protokoll 17 zu implementieren?

softwarecrash, 2024-03-01:
Ja da muss noch eine Menge angepasst werden, schau dir mal dazu zb qpiri Datei. Jede Datei davon hat eine Aufgabe und Funktionen die abgearbeitet werden um die Daten aufzubeteiten

softwarecrash, 2024-03-02:
mit welcher baudrate arbeitet das p17 überhaupt?

Steef-the-Beef, 2024-03-06:
9600 - 8 - N - 1

softwarecrash, 2024-03-06:
Thats why you dont get Data, you need to modify the autodetect function to Swap the Speed to 9600, pi30 and 18 have 2400 on some devices.
In the autodetect function it is a var to Set this