softwarecrash/Solar2MQTT-ESP8266 #64: [Bug]: Why information from Inverter not showing up just one blink and missing

rkoczen, 2024-01-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Information from Inverter not showing up just for a moment and missing

### Screenshots / Fotos

[image]

### Steps To Reproduce

First version is showing data

### Version

1.x.x and above

### Relevant livejson output

```json
{"EspData":{"Device_name":"Solar2MQTT","ESP_VCC":3.032,"Wifi_RSSI":-65,"sw_version":"1.1.6_Dongle","Free_Heap":22416,"HEAP_Fragmentation":11,"json_memory_usage":259},"DeviceData":{},"LiveData":{}}
```

### What browsers are you seeing the problem on?

Chrome

softwarecrash, 2024-01-11:
Hello,
What you mean with First Version Work? What for a First Version?
What for a dongle you use?
What for Inverter? Protocol?

Need more information about you Setup and at wich Point of Change it Stop working.

rkoczen, 2024-01-11:
- V0.5.1 
- wemos d1 mini
- easun smh 3k

with this version it works:)
yes i know about mismatch but i was try all versions :)

rkoczen, 2024-01-12:
ok i'll stay on old version because it is works:)

closed 2024-01-12 by softwarecrash