softwarecrash/Solar2MQTT-ESP8266 #65: New working easun igrid sv iv 5.6 kW device

amorfius, 2024-01-16:
### What device are you using?

Good afternoon, respect for your work.

I have a hybrid inverter easun igrid sv iv 5.6 kW (which can mix electricity) I can’t make an animation for it. 

sensor.solar2mqtt_pv_charging_power
sensor.solar2mqtt_ac_out_watt

[image]

AC(inverter&grid) input Watt
how to get data for the Power flow card plus and Home Asistant Energy card

AC(inverter&grid) input Watt or sensor.solar2mqtt_ac_out_watt - sensor.solar2mqtt_pv_charging_power

amorfius, 2024-01-16:
approximately this is what you need 
[image]

softwarecrash, 2024-01-16:
Thank you for reporting a working Device.

closed 2024-01-16 by softwarecrash

Jarnsen, 2024-01-17:
Some hybrid inverter does not provide charged and discharged energy sensors. But we can do this ourselves very quickly. Take a look in the daly2mqtt Wiki, there is your solution.