softwarecrash/Solar2MQTT-ESP8266 #68: Sola2mqtt auto discovery

darkazar, 2024-02-04:
### A clear and concise description of what the problem is.

I have mosquito broker installed on HA and multiple devices connected. Ha discovery is enabled. 
When Solar2mqtt is connected to broker, mqtt explorer shows data but its not discovered in HA.  I think HA discovery option in Solar2mqtt is not working. I have tested with second instance of HA and result is the same, no discovery. 
But in Daly2mqtt that i have too, autodiscovery is working fine ...

[image]

softwarecrash, 2024-02-04:
Looks like you get no data from your device, i see only the esp data.
please provide the esp-ip/livejson data

darkazar, 2024-02-04:
Yes , i figured it out. I have broken max 3232. 
Solved. Thank You.

closed 2024-02-04 by softwarecrash