softwarecrash/Solar2MQTT-ESP8266 #74: Intermittent connection

dotumair, 2024-03-04:
### A clear and concise description of what the problem is.

Hi, I'm facing intermittent connectivity in NodeMCU ESP8266. I've followed the wiring diagram from Wiki. Sometimes my NodeMCU picks up the values from my inverter and most of the time it fails, like no data. 

I'm using this inverter
[Crown Xavier 3.6KW](https://crownmicroglobal.com/product/crown-3-6-kw-inverter)
It's voltronic based as I can connect it to my PC using watchpower software
One thing to notice is that my inverter is sending data over 6,7 volts and when connected RS-232 to TTL it comes around 5v

I've attached the logs from watchpower software. 

Below is the JSON data that I've recorded once I am able to connect
`{
   "EspData":{
      "Device_name":"Solar2MQTT",
      "ESP_VCC":3.017,
      "Wifi_RSSI":-70,
      "sw_version":"1.1.6",
      "Free_Heap":24928,
      "HEAP_Fragmentation":9,
      "json_memory_usage":1271
   },
   "DeviceData":{
      "AC_in_rating_voltage":230,
      "AC_in_rating_current":15.6,
      "AC_out_rating_voltage":230,
      "AC_out_rating_frequency":50,
      "AC_out_rating_current":15.6,
      "AC_out_rating_apparent_power":3600,
      "AC_out_rating_active_power":3600,
      "Battery_rating_voltage":24,
      "Battery_re-charge_voltage":23,
      "Battery_under_voltage":21,
      "Battery_bulk_voltage":28.2,
      "Battery_float_voltage":27,
      "Battery_type":"User",
      "Current_max_AC_charging_current":10,
      "Current_max_charging_current":20,
      "Input_voltage_range":"Appliance",
      "Output_source_priority":"Solar first",
      "Charger_source_priority":"Only solar charging permitted",
      "Parallel_max_num":1,
      "Machine_type":"Off Grid",
      "Topology":"Transformerless",
      "Output_mode":0,
      "Battery_re-discharge_voltage":27,
      "PV_OK_condition_for_parallel":0,
      "PV_power_balance":1,
      "Device_Model":"VMII-3600",
      "Protocol_ID":"PI30"
   },
   "LiveData":{
      "AC_in_Voltage":234.3,
      "AC_in_Frequenz":49.7,
      "AC_out_Voltage":234.3,
      "AC_out_Frequenz":49.7,
      "AC_out_VA":0,
      "AC_out_Watt":0,
      "AC_out_percent":0,
      "Inverter_Bus_Voltage":334,
      "Battery_Voltage":24.4,
      "Battery_Charge_Current":0,
      "Battery_Percent":35,
      "Inverter_Bus_Temperature":35,
      "PV_Input_Current":0,
      "PV_Input_Voltage":0,
      "Battery_SCC_Volt":0,
      "Battery_Discharge_Current":0,
      "Status_Flag":10100,
      "Battery_voltage_offset_fans_on":0,
      "EEPROM_Version":0,
      "PV_Charging_Power":3,
      "Device_Status":10,
      "Solar_feed_to_Grid_status":0,
      "Country":1,
      "Solar_feed_to_grid_power":0,
      "Battery_Load":0,
      "PV_Input_Power":0,
      "Inverter_Operation_Mode":"Line",
      "Time_until_absorb_charge":0,
      "Time_until_float_charge":0,
      "Tracker_temperature":0,
      "Inverter_temperature":27,
      "Battery_temperature":35,
      "Transformer_temperature":25,
      "Fan_speed":30,
      "Inverter_charge_state":"Bulk stage"
   }
}`

Kindly help as I'm stuck in this

dotumair, 2024-03-04:
[watchpower logs.zip](https://github.com/softwarecrash/Solar2MQTT/files/14475759/watchpower.logs.zip)

dotumair, 2024-03-04:
[image]

I think it's cheap version

also serial try to detect inverter protocol at boot and fails 3 times.

dotumair, 2024-03-04:
After some reboot tries it detects the inverter and protocol is PI30.

softwarecrash, 2024-03-04:
when it found the protocol the esp run normal, then you can check the serial/webserial for error log?

dotumair, 2024-03-05:
When it finds the protocol it starts getting data and working fine.. everything works perfectly. The problem is when I restart or power off the nodemcu the communication is lost. After some reboot tries it again gets connected. Like there is no reliability of the connection.

softwarecrash, 2024-03-05:
strange, normaly the autodetect works fine.

can you show up your wireing / setup ? 
the most inverters have a power line on the rj, make sure there is nore shorting that the voltage drops

closed 2024-04-19 by github-actions[bot]