softwarecrash/Solar2MQTT-ESP8266 #77: [Bug]: HA is not subscribed to LWT

negronoir, 2024-03-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Hello,

I've just build the adaptor and installed Solar2MQTT v1.1.6

First of all: thanks for the project, it works great.

The issue I've detected is that on HA discovery, LWT (Will message) is not included, so HA doesn't subscribe to Solar2MQTT LWT topic.

Checking MQTT broker logs, I can see that LWT message is received at connection

```
2024-03-14 16:48:27: New client connected from 192.168.1.194:54937 as Axpert-0DD700 (p2, c1, k15).
2024-03-14 16:48:27: Will message specified (5 bytes) (r1, q0).
2024-03-14 16:48:27: 	Axpert/Alive
2024-03-14 16:48:27: Sending CONNACK to Axpert-0DD700 (0, 0)
2024-03-14 16:48:27: Received PUBLISH from Axpert-0DD700 (d0, q0, r1, m0, 'Axpert/Alive', ... (4 bytes))
```

but after that HA does not subscrite to it.

The rest of devices connected to the same broker subscribes HA to their LWTs

When inverter comms fails, I can check that MQTT broker detects timeout but LWT message is not forwarded to HA and latest received values remain in HA forever. Never changes to "Unavailable"

```
2024-03-14 16:56:21: Client SmartEVSE-Mercedes has exceeded timeout, disconnecting.
2024-03-14 16:56:21: Sending PUBLISH to mqtt_home_assistant (d0, q0, r0, m0, 'SmartEVSE-Mercedes/connected', ... (7 bytes))
2024-03-14 16:56:21: Client Axpert-0DD700 has exceeded timeout, disconnecting.   <--- NOT PUBLISHED IN HA
2024-03-14 16:56:21: Client DVES_AA0918 has exceeded timeout, disconnecting.
2024-03-14 16:56:21: Sending PUBLISH to mqtt_home_assistant (d0, q0, r0, m0, 'tele/tasmota_AA0918/LWT', ... (7 bytes))
2024-03-14 16:56:21: Client SmartEVSE-Tesla has exceeded timeout, disconnecting.
2024-03-14 16:56:21: Sending PUBLISH to mqtt_home_assistant (d0, q0, r0, m0, 'SmartEVSE-Tesla/connected', ... (7 bytes))
```

Thanks!

P.S. I have an Axpert MKSIV 5600. The integration detects an MKS2-5600 through protocol PI30 and does not decode all sensors and commands but at least it does with the most important ones. If I can help to update your templates with the full set of my inverter parameters, let me know how can I do.

### Screenshots / Fotos

None

### Version

1.x.x and above

### Relevant livejson output

```json
None
```

### What browsers are you seeing the problem on?

no Issue with the Browser or WebUI

softwarecrash, 2024-03-16:
hello,
there is only the alive datapoint, this is the LWT.
so i dont know how to pass this to HA that is used for the healtcheck

softwarecrash, 2024-03-18:
@negronoir 
can you check this?
[Solar2MQTT_d1_mini_1.1.6A2_debug.zip](https://github.com/softwarecrash/Solar2MQTT/files/14635871/Solar2MQTT_d1_mini_1.1.6A2_debug.zip)

negronoir, 2024-03-19:
Like a charm. Now works :-)

2024-03-19 08:52:22: Client SmartEVSE-Mercedes has exceeded timeout, disconnecting.
2024-03-19 08:52:22: Sending PUBLISH to mqtt_home_assistant (d0, q0, r0, m0, 'SmartEVSE-Mercedes/connected', ... (7 bytes))
**2024-03-19 08:52:28: Client Axpert-0DD700 has exceeded timeout, disconnecting.
2024-03-19 08:52:28: Sending PUBLISH to mqtt_home_assistant (d0, q0, r0, m0, 'Axpert/Alive', ... (5 bytes))**
2024-03-19 08:52:28: Client SmartEVSE-Tesla has exceeded timeout, disconnecting.
2024-03-19 08:52:28: Sending PUBLISH to mqtt_home_assistant (d0, q0, r0, m0, 'SmartEVSE-Tesla/connected', ... (7 bytes))
2024-03-19 08:52:46: Client DVES_DDE976 has exceeded timeout, disconnecting.
2024-03-19 08:52:46: Sending PUBLISH to mqtt_home_assistant (d0, q0, r0, m0, 'tele/tasmota_DDE976/LWT', ... (7 bytes))

Thanks!

softwarecrash, 2024-03-19:
great, will come up official with the netxt release

closed 2024-03-19 by negronoir