softwarecrash/Solar2MQTT-ESP8266 #80: Solar2MQTT V0.5.1 WORK, V1.1.6 NO

Konycz, 2024-03-24:
### A clear and concise description of what the problem is.

Hi, I'm running Solar2MQTT V0.5.1 on my D1 mini and everything works, but as soon as I install the latest version 1.1.6, the device doesn't tell me the values from the inverter
[image]
[image]

Konycz, 2024-03-24:
{
  "EspData": {
    "Device_name": "Solar2MQTT",
    "ESP_VCC": 3.013,
    "Wifi_RSSI": -46,
    "sw_version": "1.1.6_debug d1_mini Dec 30 2023 15:05:55",
    "Free_Heap": 24320,
    "HEAP_Fragmentation": 3,
    "json_memory_usage": 225,
    "json_capacity": 2048,
    "runtime": 9,
    "ws_clients": 0
  },
  "DeviceData": {},
  "LiveData": {}
}

Konycz, 2024-03-24:
EASUN POWER 3KW 230VAC 24V Soalr Inverter 50Hz 60Hz 3000VA PWM 70A Pure Sine Wave Charge Current Inversor With Battery Charge

rabeno, 2024-03-24:
Hi,
From the release page (https://github.com/softwarecrash/Solar2MQTT/releases) it seems that from version 1.0 there was a wiring change, so 1st thing to do is to check if your wiring is correct according to the wiring diagram (https://github.com/softwarecrash/Solar2MQTT/wiki/Wiring-Diagram). Hope this helps

Konycz, 2024-03-24:
yes, you were right, thank you very much, but it displays the temperature quite strangely
[image]

rabeno, 2024-03-24:
Yes, I have the same problem, but according to this (https://github.com/softwarecrash/Solar2MQTT/issues/72#issuecomment-1935013899), there is nothing to do, since this software cannot translate the numeric value to real temperature.

rabeno, 2024-03-24:
BTW, I have been investigating this issue myself, and according to the protocol description, the value of the temperature being displayed by this firmware should be divided by 10 in order to get the real temperature in Celsius. In your case, the real temp is 58.2

closed 2024-05-09 by github-actions[bot]