softwarecrash/Solar2MQTT-ESP8266 #81: EASUN POWER SMG-II-3.6K-WIFI

Camposmansi, 2024-03-25:
### A clear and concise description of what the problem is.

I have this inverter and it does not work, what is needed so that it can be included? I am asking the manufacturer for the protocol

Thank you so much

softwarecrash, 2024-03-25:
Hello,
Work your Inverter with watchpower or Solarpower Software (PC)? Please Check this with your Manual or seller

Camposmansi, 2024-03-27:
I couldn't get it to work.

Work with a WiFi module for example: https://ae01.alicdn.com/kf/S964b7336e20f42a090f4163013cfc92aY/EASUN-M-dulo-WiFi-con-soluci-n-de-monitoreo-remoto-RS232-dispósito-inal-mbrico- for-investor.jpg_.webp

Use the dessmonitor website or the SmartESS APP

softwarecrash, 2024-03-27:
That was Not the question.

Sure that your wireing and other things works well?

Camposmansi, 2024-03-28:
> That was Not the question.
> 
> Sure that your wireing and other things works well?

Yes, I have tried with 2 different TTL MAX, I have asked EASUN about the Protocol, I attached it, to see if it can come.
[SMG-RS232 Communication Protocol V1.0(1).pdf](https://github.com/softwarecrash/Solar2MQTT/files/14785967/SMG-RS232.Communication.Protocol.V1.0.1.pdf)

softwarecrash, 2024-03-28:
thank you, your device uses modbus as communication, this is not implemented at the moment, its on the ToDo list, but unknown when i find time to implement this communication form, its complete different from the PiXX protocols

Camposmansi, 2024-03-28:
I cry :'( 

Thank you very much for everything, I will look forward to it.

mihaigsm2003, 2024-04-03:
when i received one SMG inverter to repair will try do some software for that inverter

closed 2024-05-18 by github-actions[bot]

vGubriienko, 2024-10-10:
Any updates about SMG II Invertors support? I could help coding/debug if you point me where I could start.

softwarecrash, 2024-10-10:
Modbus is now integrated in pre Release, but we need the Registers.
@bigdigital need you here

vGubriienko, 2024-10-10:
I've tried [V1.2.0-Pre3](https://github.com/softwarecrash/Solar2MQTT/releases/tag/V1.2.0-Pre3) and get next logs, does it OK?
```
[WebSerial] Connecting...
[WebSerial] Connected...
[C:   QPI][CR:    0][CC:    0][L:   3]

QPI:		NAK (Length: 3)

[C: ^P005PI][CR:    0][CC:    0][L:   3]

^P005PI:		NAK (Length: 3)

Try Autodetect Protocol

[C:   QPI][CR:    0][CC:    0][L:   3]

QPI:		NAK (Length: 3)

[C: ^P005PI][CR:    0][CC:    0][L:   3]

^P005PI:		NAK (Length: 3)

Try Autodetect Protocol

[C:   QPI][CR:    0][CC:    0][L:   3]

QPI:		NAK (Length: 3)

[C: ^P005PI][CR:    0][CC:    0][L:   3]

^P005PI:		NAK (Length: 3)

Try Autodetect Modbus device

Response Timed Out

Response Timed Out

Time-out

Request failed!

Response Timed Out

Response Timed Out

Time-out

Request failed!

Response Timed Out

Response Timed Out

Time-out

Request failed!

Response Timed Out

Response Timed Out

Time-out

Request failed!

----------------- End Autodetect -----------------

ProzessData P:0 C:0
ProzessData P:0 C:0

```

softwarecrash, 2024-10-20:
no thats not ok, it tells it cant get data