softwarecrash/Solar2MQTT-ESP8266 #92: Web interfaces

Konycz, 2024-05-17:
### Is your feature request related to a problem? Please describe

Is there any possibility to grant a www interface?  It's about: I have a device in the garden and I want to control the switch off via the arduino according to the state of the batteries, i.e. when it has a 25V battery, the switch will turn on, as soon as it has a 27V battery, the switch will turn off.  It would be enough if the device had a web interface where I could load the current values.  Now I have it through HA and tuya, but what if the internet turns off, it wouldn't work.

### Describe the solution you'd like

. 

### Describe alternatives you've considered

. 

### Additional context

.

softwarecrash, 2024-05-18:
Please Take a Look at the wiki for possible ways to Set or get data

Konycz, 2024-05-18:
Where exactly?  I guess I'm still overlooking it :)

softwarecrash, 2024-05-18:
https://github.com/softwarecrash/Solar2MQTT/wiki/

Konycz, 2024-05-18:
Yes, I know where the wiki is, but where and how to set it up so that I can read the data from the second device on the local network, when the mqtt server is elsewhere and when the internet goes down, I can't access it.  I really need to read the values ​​even when the internet is not working

softwarecrash, 2024-05-18:
When Internet Not working but Network avaible you can get data over esp-ip/livejson, it Delivery all Data as json String.
Unfortunaly is See its Not well documented in this project

Konycz, 2024-05-18:
And how can this data be obtained?

softwarecrash, 2024-05-18:
Send from another ESP or Something Network connected a http request to this Adress.

Konycz, 2024-05-18:
I don't understand this... could I ask for an explanation?? what will the address look like to which I will connect from the second device to get the raw data in json??

softwarecrash, 2024-05-18:
The Adresse ist http://esp-ip/livejson

For Test Just call this Adress in your Browser, you will get the livejson String.
So with any other connected device you need to build a http request.
Check 
https://www.arduino.cc/reference/en/libraries/httpclient/ for more details

Konycz, 2024-05-18:
Great, thank you very much