softwarecrash/Solar2MQTT-ESP8266 #95: [HELP] Please add support for the MUST inverter ph1800 Model 3024 plus

Spais07, 2024-06-05:
### A clear and concise description of what the problem is.

I have MUST inverter ph1800 Model 3024 plus
RJ45 with RS485 interface.
I tried to create a converter but it doesn't work with my inverter.
Although its performance is similar to those on which these converters work.
I made a request to the developer and they sent me the file with RS485 Modbus RTU communication Protocol (Attached here)
And other information that I have
[PH1800 PV1800 EP1800 PV3500 EP3500 RS485 Modbud RTU communication Protocol1.4.15&nbsp_ 20180901.xlsx](https://github.com/user-attachments/files/15586297/PH1800.PV1800.EP1800.PV3500.EP3500.RS485.Modbud.RTU.communication.Protocol1.4.15.nbsp_.20180901.xlsx)
[must_ph18_3024_plus_manual.pdf](https://github.com/user-attachments/files/15586303/must_ph18_3024_plus_manual.pdf)
[must_ph18_3024_plus_pasport.pdf](https://github.com/user-attachments/files/15586304/must_ph18_3024_plus_pasport.pdf)
[must_ph18_3024plus_datasheet.pdf](https://github.com/user-attachments/files/15586307/must_ph18_3024plus_datasheet.pdf)

### Additional context

In addition, I want to say that I managed to connect the inverter to the Home Assistant via the USB-B cable, but the connection is unstable.
The integration can work for 2-3 hours and stop collecting data from the inverter until the Home Assistant is completely restarted.
Here is link on that integration:
https://github.com/mukaschultze/ha-must-inverter

softwarecrash, 2024-06-05:
Hello,
Modbus communication is actual Not supported.

closed 2024-07-20 by github-actions[bot]

softwarecrash, 2024-10-02:
now we have support for MUST devices in pre release