softwarecrash/Solar2MQTT-ESP8266 #104: Powmr 1.6kw not working in wemos and no data i hope this fix a soon

Ching858, 2024-08-05:
### What device are you using?

Powmr 1.6kw not working I can't access the data.. I hope my problem will fix soon. Please see the attached file.
Thank you
[image]

derLoosi, 2024-08-05:
Does your device work using the Watchpower PC Software? I cant see your Powmr 1.6kw on the [Working devices list](https://github.com/softwarecrash/Solar2MQTT/wiki/Confirmed-Working-Device-List)

hvdb, 2024-08-05:
I have the same one.
Although it's rated as 1.6kw

Type is pow-hvm2.0h-12v

Doesn't work for me as well, can t find the controller it seems

I think I have it wired correctly

[image]
[image]
[image]

softwarecrash, 2024-08-06:
Again...
Please Check if the device Work with the PC Software watchpower or Solarpower, when this will Work, the ESP Firmware will Work too.
Actual only the communication from this sowares are supported.

Ching858, 2024-08-10:
What is watch power or solarpower

Ching858, 2024-08-10:
> Again... Please Check if the device Work with the PC Software watchpower or Solarpower, when this will Work, the ESP Firmware will Work too. Actual only the communication from this sowares are supported.

where can I download Solarpower?

Ching858, 2024-08-10:
> Again... Please Check if the device Work with the PC Software watchpower or Solarpower, when this will Work, the ESP Firmware will Work too. Actual only the communication from this sowares are supported.

I used wifi module via smartess it's works but when it comes to wemos it doesn't

Ching858, 2024-08-10:
[image]

Ching858, 2024-08-10:
[image]

Ching858, 2024-08-11:
> What do you not understand about the statement that if it works with smartess, it does **NOT** work with solar2mqtt? And that's exactly what your screenshot from the webserial says: **NAK**! The Wemos gets no response from the inverter. How can it, if the protocol does NOT fit on both sides?

So what im going to do?

softwarecrash, 2024-08-11:
> So what im going to do?

Nothing, my Software Doenst Work with ,our device