softwarecrash/Solar2MQTT-ESP8266 #116: [Bug]: PV2 not monitored on  Daxtromn AGH-10.2KW (analog VICTOR NM-MAX-10.2KWs)

fastvd, 2024-09-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Greetings. thanks for the great program….
I have a VICTOR NM-MAX-10.2KWs inverter, it works based on the PI30 protocol. this inverter has 2 PV...the problem is that Solar2Mqtt reads only PV1...it also has a wifi dongle that shows data in the smartESS application, there is data from PV1 and PV2...so the question is how to read or OVERALL solar generation , or separately PV1 and PV2

### Screenshots / Fotos

[image]
[image]

### Version

1.x.x and above

### Relevant livejson output

```json
{"EspData":{"Device_name":"Solar2MQTT","ESP_VCC":3.074,"Wifi_RSSI":-38,"sw_version":"1.1.7","Free_Heap":23704,"HEAP_Fragmentation":8,"json_memory_usage":942,"json_capacity":2048,"runtime":73,"ws_clients":0,"detect_protocol":2},"DeviceData":{"Device_Model":"VMII-NXPW5KW","Protocol_ID":"PI30"},"LiveData":{"AC_in_Voltage":244.2,"AC_in_Frequenz":49.9,"AC_out_Voltage":220.3,"AC_out_Frequenz":49.9,"AC_out_VA":4280,"AC_out_Watt":4265,"AC_out_percent":46,"Inverter_Bus_Voltage":465,"Battery_Voltage":55.4,"Battery_Charge_Current":27,"Battery_Percent":72,"Inverter_Bus_Temperature":42,"PV_Input_Current":8.2,"PV_Input_Voltage":414.7,"Battery_SCC_Volt":0,"Battery_Discharge_Current":0,"Status_Flag":10110,"Battery_voltage_offset_fans_on":0,"EEPROM_Version":0,"PV_Charging_Power":3336,"Device_Status":110,"Battery_Load":27,"PV_Input_Power":3312,"Inverter_Operation_Mode":"Battery","Time_until_absorb_charge":1,"Time_until_float_charge":0,"Tracker_temperature":0,"Inverter_temperature":32,"Battery_temperature":17,"Transformer_temperature":42,"Fan_speed":100,"Inverter_charge_state":"Float"}}
```

### What browsers are you seeing the problem on?

Chrome

softwarecrash, 2024-09-17:
hello,
without the communication protocol its not possible to implement more than one pv string

fastvd, 2024-09-17:
> hello, without the communication protocol its not possible to implement more than one pv string

[image]

fastvd, 2024-09-17:
!!! PI30 !!!
it was visible even on the 1st screen

softwarecrash, 2024-09-17:
Stop using hordening Symbols, its Like angry...

So the PI30 Protocol is Not one single Protocol, it includes so i know more than 10 subversions, and much more variants, so for that what you need, i need a complete Protocol Snipet and Datasheet for it.

That all is Not easy as sound

fastvd, 2024-09-18:
I'm sorry if you thought I was angry and wanted to offend someone) in fact, I just want to understand how my inverter works and read all the data I need.  now to the point...then how do I find out which protocol I have?  maybe there are some sniffers or just firmware that polls different protocols or take Wireshark and scan my existing wifi dongle?

fastvd, 2024-09-18:
> have the device a USB port? when yes, you can connect it with a PC, it will emulate a serial port, so you can use watchpower or solarpower software for monitoring and settings of the device, when this will work, you can enable the debug function of this software, this will write a bunch of log files, that what i need to see what the device talks.
> 
> for explaining, take a look at the documents here: https://github.com/softwarecrash/Solar2MQTT/tree/master/Protocol
> 
> there are more than one protocol of the PI30, and many different versions of command and answer strings, so we need to find out how to ask your device for the second PV string data. in some protocols are a second string implemented, and in my firmware a function try to get this data, so its possible that your device need another request or send a different answer.

[image]
yes, I have usb-rs232 and I can read and change data... but there is no info on the protocol... and I don't see where in the menu you can enable debugging...

fastvd, 2024-09-18:
according to the link you gave, there are simply descriptions of various protocols, but there are no instructions on how to separate it... maybe I'm not looking there?

softwarecrash, 2024-09-18:
The debug is Hidden, right click on the Icon in the taskbar and select enable debug

fastvd, 2024-09-18:
I found a program similar to watchpower.. this one http://www.multisibcontrol.net/ it looks like a lot more debug functions
[image]
[image]

fastvd, 2024-09-18:
> > yes, I have usb-rs232 and I can read and change data... but there is no info on the protocol... and I don't see where in the menu you can enable debugging...
> 
> You can enable it just in the tray icon (near the clock on your taskbar). ![image](https://private-user-images.githubusercontent.com/17761850/368659929-24b646ec-22a9-44dd-b27f-dfa95688b2b8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY2NzgxNjYsIm5iZiI6MTcyNjY3Nzg2NiwicGF0aCI6Ii8xNzc2MTg1MC8zNjg2NTk5MjktMjRiNjQ2ZWMtMjJhOS00NGRkLWIyN2YtZGZhOTU2ODhiMmI4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTE4VDE2NDQyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE2Y2E5YjRlMmRmZTM2MWQ4YWQ3ZDJjZDUxNjYzMmI2Mjk0MmE3NDRlMWQ1MTYyNTE1M2U0MGMwYWU4NTg2YTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.JSSS58zY9YVv37Ox__8RrZh243NiMY9L_FrM6ixEj8Q)

[image]
I don't have debug mode ((((

fastvd, 2024-09-18:
but with MultiSIBcontrol I can directly run different commands...
[image]

fastvd, 2024-09-18:
this app also doesn't see PV2(((

I started QPIGS2 a few days ago directly from mqtt...but I didn't get a response then...the only thing I see is PV2 - smartESS via wifi dongle

fastvd, 2024-09-18:
give links to watchpower where there is a debug mode

fastvd, 2024-09-18:
> > give links to watchpower where there is a debug mode
> 
> https://www.mppsolar.com/v3/download/ under Monitoring Software

[image]

I downloaded a newer version, a floating window appeared, where the debug mode is checked immediately, the folder where the logs should be created opens immediately, but the logs themselves do not appear (((maybe it takes some time?

fastvd, 2024-09-18:
25 minutes = 0 in debug(((

softwarecrash, 2024-09-18:
Disconnect and reconnext the device

fastvd, 2024-09-19:
[image]

fastvd, 2024-09-19:
https://github.com/odya/esphome-powmr-hybrid-inverter

fastvd, 2024-09-19:
+ you need this

 - platform: modbus_controller
    modbus_controller_id: smg_inverter
    name: "PV Voltage2"
    id: pv_voltage2
    address: 4563
    register_type: holding
    value_type: U_WORD
    unit_of_measurement: "V"
    device_class: voltage
    state_class: measurement
    accuracy_decimals: 1
    lambda: |-
      return swapBytes(x);
    filters:
      - multiply: 0.1

  - platform: modbus_controller
    modbus_controller_id: smg_inverter
    name: "PV Power2"
    id: pv_power2
    address: 4564
    register_type: holding
    value_type: U_WORD
    unit_of_measurement: "W"
    device_class: power
    state_class: measurement
    accuracy_decimals: 1
    lambda: |-
     return swapBytes(x);

softwarecrash, 2024-09-19:
> * you need this
>
> * platform: modbus_controller
>   modbus_controller_id: smg_inverter
>   name: "PV Voltage2"
>   id: pv_voltage2
>   address: 4563
>   register_type: holding
>   value_type: U_WORD [...]

thats complete wrong....
this is modbus with registers!
my software read rs232 serial with qpigs commands, modbus not supprted atm

fastvd, 2024-09-19:
I understand that you read the data in a different way...I just showed how I solved my problem and maybe you will take something useful for yourself to expand the functionality of your wonderful product!

fastvd, 2024-09-19:
[image]
this piece of code I pasted into the author's project to read PV2

fastvd, 2024-09-19:
the data I see in HA matches what I see physically on the inverter display! this is what i needed!

softwarecrash, 2024-09-19:
modbus register support are planed, but need a lot of time that i dont have yet

closed 2024-09-29 by softwarecrash

VladimirButjko, 2024-11-02:
Tell me please, is it planned to fix the display of information on the second field of solar panels? On the inverter PowMr 10.2 exactly the same situation. Found a similar problem that was solved by changing the length of the request QPIGS2
https://github.com/manio/skymax-demo/issues/23
Maybe here it can be fixed in the same way.