softwarecrash/Solar2MQTT-ESP8266 #124: [HELP] I am not receiving data

IvanSerevko, 2024-09-18:
### A clear and concise description of what the problem is.

### Please enter the name of the device manufacturer
PowMr

### Please enter the device type
6.2 kW

### Used Hardware
Wemos D1 Mini
rs232 to ttl
dc-dc 12 to 5v

Hello, I did everything according to the instructions, but I have a problem with receiving data after connecting via mqtt, when I configure the connection to mqtt, the data is not displayed, and also in mqtt explorer I see my inverter but do not see its data, what could be the problem, please help me

vytya, 2024-09-18:
I don't see any data either on PowMr 6.2.
BTW I tried couple esp32 solutions and also without success. 

But powmr wifi module is working well.

IvanSerevko, 2024-09-18:
same situation

softwarecrash, 2024-09-18:
hey
have one of our devices a serial port or usb port that allow communication with a pc? so that we can try to get the communication between the pc software and the device?

IvanSerevko, 2024-09-18:
rs232 for wifi  and rs485 for bms

softwarecrash, 2024-09-18:
you can try to use a rs232 cable or a adapter rs232-usb and connect it with a pc, and use watchpower or solarpower, hope you get data from the device

vytya, 2024-09-18:
[image]
Yee! It started to working. 
As it turned out I connected to wrong rs232 pins.

@IvanSerevko можеш написати якщо що в тг @vytya

softwarecrash, 2024-09-18:
can it close?

IvanSerevko, 2024-09-18:
Yes, I will try other components to solve this problem

closed 2024-09-18 by IvanSerevko