softwarecrash/Solar2MQTT-ESP8266 #125: [HELP] <title>Esp 01 Pin

aaqibraaza, 2024-09-18:
### A clear and concise description of what the problem is.

Hi . thanks for the great project, please confirm the pins for ESP01 adapter or NodeMCU if we use these instead of Wemos D1

softwarecrash, 2024-09-18:
if you use the D1 Mini image, use the D6/D7
if you use the ESP01 image use RX/TX

aaqibraaza, 2024-09-19:
Thank you very much, your prompt response is highly appreciated.

softwarecrash, 2024-09-19:
no problem, btw the dongle version is the same pinout as the ESP01 but with aditional led states

aaqibraaza, 2024-09-20:
Hi, hope you will be doing well, as i said after building this project i will let you know. well project goes successful any how i have some issue on the main screen i am not getting any value screenshot attached. But in web serial view and on mqtt explorer i am getting the real time data. project is built with ESP01.
[image]

softwarecrash, 2024-09-20:
Please Copy the content of http://esp-ip/livejson Here
So some little Things goes wrong

aaqibraaza, 2024-09-23:
well thanks for the reply and sorry late reply from my side, actually it started to work at its own , i did not changed any settings and after i restarted the mqtt server and ESP01, values comes alive on the main screen.

aaqibraaza, 2024-09-25:
Hi hope you will be doing well, i am using the project from last few days and everything is working fine well i need some help of yours regarding Energy data, i have checked http://esp-ip/livejson to find out the available sensors data but i have not found any energy related data i.e. how much energy is produced on a specif day or during a specific time interval please guide me regarding this.

softwarecrash, 2024-09-25:
> Hi hope you will be doing well, i am using the project from last few days and everything is working fine well i need some help of yours regarding Energy data, i have checked http://esp-ip/livejson [...]

The Date comes all from the device, when this Doenst have acumulated Data you dont See it.
What you get is deepend in the device.
Otherwise you need to calculate it self

aaqibraaza, 2024-09-25:
ok i got the point. to make it simple suppose my device is connected at all the time let say for a month then what formula should i use  and which topics of mqtt will be included in the calculations. i am asking for this because you have already did it in the main.h file
    {"PV_generation_day", "solar-power-variant", "Wh", "energy"},
    {"PV_generation_month", "solar-power-variant", "Wh", "energy"},
    {"PV_generation_sum", "solar-power-variant", "Wh", "energy"},
    {"PV_generation_year", "solar-power-variant", "Wh", "energy"},

softwarecrash, 2024-09-25:
> ok i got the point. to make it simple suppose my device is connected at all the time let say for a month then what formula should i use and which topics of mqtt will be included in the calculations. i am asking for this because you have already did it in the main.h file {"PV_generation_day", "solar-power-variant", "Wh", "energy"}, {"PV_generation_month", "solar-power-variant", "Wh", "energy"}, {"PV_generation_sum", "solar-power-variant", "Wh", "energy"}, {"PV_generation_year", "solar-power-variant", "Wh", "energy"},
> 
> i am not using home-assistant just want to develop my own dashboard with mqtt published topics. so please guide me accordingly

no, this "lists" are for HomeAssistant discovery, and for devices that have the values.
when you dont find the acumulated data in your mqtt, your device doenst have this. so you need to collect it by itself. there is no calculation at the esp

aaqibraaza, 2024-09-25:
Ok i got the point, thank you very much for your time.

closed 2024-09-29 by softwarecrash