softwarecrash/Solar2MQTT-ESP8266 #135: [HELP] How to update Wifi settings?

EOUpSL93Y, 2024-09-28:
### A clear and concise description of what the problem is.

Hello. Is there any way to update wifi connection settings?

softwarecrash, 2024-09-28:
or bring the esp out of range of the old wifi, after a short time it will open the AP mode

EOUpSL93Y, 2024-09-28:
1. It's a ESP01;
2. I have access to web interface, so i can press "Reset ESP". But i don't want to reconfigure everything again, i just want to change AP ESSID to another (with stronger signal).

EOUpSL93Y, 2024-09-28:
It would be nice to have such setting.

EOUpSL93Y, 2024-09-28:
> As I said, not possible at the moment.

Ok.

> BTW: Two different Wifi's in the house?

1 in the house as a main AP, 2nd for other locations (close to house) first of all for "smart" modules.
So, for example, in ESP Home i have 2 wifi definitions in device config, i can configure/debug it at home and then install in any other location.

closed 2024-09-28 by EOUpSL93Y