softwarecrash/Solar2MQTT-ESP8266 #145: Anenji ANJ-4000-24V

Anatol72, 2024-10-08:
### Please enter the name of the device manufacturer

Anenji

### Please enter the device type

ANJ-4000-24V

Anatol72, 2024-10-08:
maybe someone has integrated this inverter

softwarecrash, 2024-10-08:
Is it working or is this a question for compatibility?

closed 2024-10-18 by softwarecrash

yaroslav2901, 2024-12-09:
I also cannot connect this inverter.  If I understand correctly, the communication protocol cannot be determined

> [WebSerial] Connecting...
[WebSerial] Connected...
[C:    QPI][CR:    0][CC:    0][L:   3]
QPI:		NOA (Length: 3)
[C: ^P005PI][CR:    0][CC:    0][L:   3]
^P005PI:		NOA (Length: 3)
Try Autodetect Protocol
[C:    QPI][CR:    0][CC:    0][L:   3]
QPI:		NOA (Length: 3)
[C: ^P005PI][CR:    0][CC:    0][L:   3]
^P005PI:		NOA (Length: 3)
Try Autodetect Modbus device
Init MUST PV/PH 18 protocol, baud 19200, modbusAddr 4
Response Timed Out
Response Timed Out
Time-out
Request failed!
Response Timed Out
Response Timed Out
Time-out
Request failed!
Init DEYE protocol, baud 9600, modbusAddr 1
Init Anenji protocol, baud 9600, modbusAddr 1
----------------- End Autodetect -----------------
ProzessData P:0 C:0
WS data send
Data sent to MQTT
----------------- Start Autodetect -----------------
Try Autodetect Protocol
[C:    QPI][CR:    0][CC:    0][L:   3]
QPI:		NOA (Length: 3)
[C: ^P005PI][CR:    0][CC:    0][L:   3]
^P005PI:		NOA (Length: 3)
Try Autodetect Protocol
[C:    QPI][CR:    0][CC:    0][L:   3]
QPI:		NOA (Length: 3)
[C: ^P005PI][CR:    0][CC:    0][L:   3]
^P005PI:		NOA (Length: 3)
Try Autodetect Protocol

`