softwarecrash/Solar2MQTT-ESP8266 #150: [FEATURE] Move some values from DeviceData section to LiveData

vpm, 2024-10-10:
### Is your feature request related to a problem? Please describe

I need some current state from futures for HA automatisation
Current_max_AC_charging_current
Output_source_priority
Charger_source_priority

### Describe the solution you'd like

Move to LiveData section and update every time when refreshing

### Describe alternatives you've considered

Hi
Cool project

vpm, 2024-10-10:
I plane using HA for charge bettery at night, but i need using charging current  depend from battery capacity, Also i dont need charging battery fulltime - at night  i have 50% discount for power price, but also i need untooched backup/ And i live in Ukraine and sometimes we also have problems with electricity

softwarecrash, 2024-10-10:
hello,
move single data to livedata is not possible.
so i moved for thest the whole static data to the automatic refresh.
try this
[Solar2MQTT_1.2.0-Pre3.1.zip](https://github.com/user-attachments/files/17332786/Solar2MQTT_1.2.0-Pre3.1.zip)

softwarecrash, 2024-10-18:
@vpm some response?

vpm, 2024-10-20:
Hi
All ok - data changed at every update

closed 2024-10-21 by vpm