softwarecrash/Solar2MQTT-ESP8266 #152: [HELP] SMG-II 2,2kw RS232

dobmeierr, 2024-10-12:
### A clear and concise description of what the problem is.

Hello,

Somebody know Easun/Isolar SMG II 2,2Kw Inverter is working?
No connection possible via rs232

Thanks

softwarecrash, 2024-10-13:
the SMG have modbus not RS232

dobmeierr, 2024-10-14:
> the SMG have modbus on RS232, support is planed

hello softwarecash,

thank you for your info.
when you somebody need for testing or something else about this product, ask me

softwarecrash, 2024-10-14:
> > the SMG have modbus on RS232, support is planed
> 
> hello softwarecash,
> 
> thank you for your info. when you somebody need for testing or something else about this product, ask me

for testing and other, youre welcome to the discord channel

dobmeierr, 2024-10-14:
> > > the SMG have modbus on RS232, support is planed
> > 
> > 
> > hello softwarecash,
> > thank you for your info. when you somebody need for testing or something else about this product, ask me
> 
> for testing and other, youre welcome to the discord channel

I just discovered that Modbus is implemented in V1.2.0-Pre3...

softwarecrash, 2024-10-14:
> I just discovered that Modbus is implemented in V1.2.0-Pre3...

modbus yes, but not the communication variant of the SMG class, i think thats the next what comes. but you can try it, not sure at the moment that smg not the same as the must or deye

dobmeierr, 2024-10-14:
> modbus yes, but not the communication variant of the SMG class, i think thats the next what comes. but you can try it, not sure at the moment that smg not the same as the must or deye

I just tried V1.2.0-Pre3 with my SMG-II, but without success. Not receiving any data.  
You say it's on your todo?

softwarecrash, 2024-10-14:
> I just tried V1.2.0-Pre3 with my SMG-II, but without success. Not receiving any data. You say it's on your todo?

yes, actual @bigdigital work on modbus i hope it build in the smg variant too

dobmeierr, 2024-10-15:
My SMG-II comes with a RS232 and a RS485 interface.
Can use both interfaces?

At the moment i've tried the RS232 without succes.

closed 2024-11-29 by github-actions[bot]