softwarecrash/Solar2MQTT-ESP8266 #154: [HELP] EASUN SMH-II-2.2 KW, 12V with an Wifi adapter that was received in the box

geoprea1989, 2024-10-13:
### A clear and concise description of what the problem is.

Hello,

I would like to ask for your help in connecting my EASUN SMH-II-2.2 KW, 12V with an Wifi adapter that was received in the box to my Lolin NodeMCU v3 board.

I seen in the Wiki that the "EASUN SMH-II-2.2KW-12V-WIFI" was confirmed as supported.

Unfortunately I don't have any D1 Mini available and I decided to give it a try with the NodeMCU clone.

I checked the PIN assignations and D6 and D7 are matching with GPIO 12 and GPIO 13 for both boards.

Regarding the RS232 to TTL converters, I tried both with the module from the Wiring diagram and with a small MAX3232 module, but there's still no output :(

The wiring I used is:

1. Recommended RS232 module scenario:

Inverter RJ-45, PIN 1 -> recommended RS232 module, PIN 3 from female DB9 connector -> recommended RS232 module, TX PIN -> Lolin NodeMCU v3, D7 PIN
Inverter RJ-45, PIN 2 -> recommended RS232 module, PIN 2 from female DB9 connector -> recommended RS232 module, RX PIN -> Lolin NodeMCU v3, D6 PIN
Inverter RJ-45, PIN 8 -> recommended RS232 module, PIN 5 from female DB9 connector -> recommended RS232 module, GND -> Lolin NodeMCU v3, GND
recommended RS232 module, VCC -> Lolin NodeMCU v3, 3.3V

2. small MAX3232 module scenario:

Photo of the module:
https://powerforum.co.za/uploads/monthly_2021_11/1578363192_microRS232(RJ45)-TTLConverter.jpg.fa93ef5e81c28febfdc615275835a342.jpg

Inverter RJ-45, PIN 1 -> small MAX3232 module, RS232 side, "->" symbol
Inverter RJ-45, PIN 2 -> small MAX3232 module, RS232 side, "<-" symbol
Inverter RJ-45, PIN 8 -> small MAX3232 module, RS232 side, GND

small MAX3232 module, TTL side, "->" symbol -> Lolin NodeMCU v3, D7 PIN
small MAX3232 module, TTL side, "<-" symbol -> Lolin NodeMCU v3, D6 PIN
small MAX3232 module, TTL side, GND -> Lolin NodeMCU v3, GND
small MAX3232 module, TTL side, VCC -> Lolin NodeMCU v3, 3.3V

I tried swapping D6/D7 pins, but there was still no output.

I tried with two NodeMCUs, two MAX3232 small modules and two recommended RS232 modules, but nothing changed.

BTW, I tried the following two firmwares:

Solar2MQTT_d1_mini_1.1.8.bin
Solar2MQTT_esp12e_1.1.8_Dongle.bin

Am I missing something?

I'd really appreciate your recommandation regarding the firmware I should use for this Lolin NodeMCU v3 board and the pinout that I should follow, both on the NodeMCU and on the recommended RS232 module and/or small MAX3232 module.

Many thanks in advance for your help! :)

geoprea1989, 2024-10-13:
Hi guys,

After a lot of struggle, I finally made it! :)

It seems that after each pin swap or change you do in the wiring, you need to reboot the module in order to autodetect the protocol.

I found this by mistake, as I rebooted the module with the USB cable still connected to my laptop while trying to flash the D1 Mini firmware by using the following URL with the Edge browser instead of Tasmotizer:
https://all-solutions.github.io/Flash2MQTT/?get=Solar2MQTT

This console output rang a bell:

"----------------- Start Autodetect -----------------
Try Autodetect Protocol"

BTW, in case it helps someone, I made it work with the recommended module, the one with the female DB9 connector soldered on.

Many thanks for creating and maintaining this project, it really helps a lot of people :D

Wish you all good luck with your green energy projects! :)

softwarecrash, 2024-10-13:
Hello,
Great that you managed IT to Work.

Yes the autodetect function Doenst Run endless at the moment, but good Point.
I will Check tonmake it Run repeadly until it found a device.

Btw: the dongle Firmware is only for reflashed WiFi dongles that was delivered with some devices.

closed 2024-10-18 by softwarecrash