softwarecrash/Solar2MQTT-ESP8266 #157: [HELP] <title> can't get proper connect/get data from POW-SunSmart 12KL3 by RS485

Nowalon, 2024-10-17:
### A clear and concise description of what the problem is.

hi! I tried to connect PowMr POW-SunSmart 12KL3 with 2 different types of RS485-to-TTL (especially with MAX485 as on the picture with same wiring to D1 mini but with the RJ45 cable pinout described here on the last page https://cdn.shopify.com/s/files/1/0746/0415/1079/files/WIFI-HF-N--RS485-_Manual.pdf?v=1715157348 (re-checked a lot)) and I'm always gettting the messages like:
`^P005PI:		NAK (Length: 3)

Try Autodetect Protocol

[C:   QPI][CR:    0][CC:    0][L:   3]

QPI:		NAK (Length: 3)

[C: ^P005PI][CR:    0][CC:    0][L:   3]

^P005PI:		NAK (Length: 3)

Try Autodetect Modbus device

Response Timed Out

Response Timed Out

Time-out

Request failed!

Response Timed Out

Response Timed Out

Time-out
`

what steps and direction would be suggested to dig in? thank you!

softwarecrash, 2024-10-18:
hello,
the messages tells that are no or wrong response from the inverter.
so i dont know your device, its possible that it is not supported at the moment by modbus, modbus is experimental now and need a lot of data and configuration.
please ask your seller or vendor for the protocol sheet like register and more

Nowalon, 2024-10-18:
hello! thank you for response.
I've found some protocol and adresses description for POW-SunSmart 12KL3  here "HomeAssistant ESPHome / RS485 Modbus Integration": https://github.com/ByteSeekerPro/PowMr-POW-SunSmart-12KL3-RS485-HomeAssistant-Integration/blob/master/docs/SRNE_ModBus_Protokoll_V1.92.pdf , https://github.com/ByteSeekerPro/PowMr-POW-SunSmart-12KL3-RS485-HomeAssistant-Integration/blob/master/src/powmr-inverter-1.yaml
but not sure I can apply this to use it with Solar2MQTT by my own.

softwarecrash, 2024-10-18:
thank you,
it needs time to implement it, modbus is new and in testing, so a lot of devices needs to build in

Nowalon, 2024-10-18:
I got it, understand. thanks again for your response. you're doing a good job with this, Solar2MQTT

closed 2024-12-02 by github-actions[bot]