softwarecrash/Solar2MQTT-ESP8266 #158: [Bug]: 1.2.0-Pre4 null values in MQTT, correct values in web

valqk, 2024-10-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Other (please write your hardware in 'What happened?')

### What happened?

Using Nodemcu v3 with pin 6,7 soldered + mas3232 mini board + crimped RJ45, attached to EASUN iGrid SV IV 5.6KW
Website works fine and values shown in web reads fine and are correct - meaning the communication with the inverter is fine.
Got MQTT in Home Assistant installed, up& running, with user/pass set. mosquitto_sub shows strings, etc.
Issue is that when I check MQTT Json Style I get no msgs in mqtt (listening for all topics). At some point it was sending
```
homeassistant/sensor/Solar01/AC_in_rating_current/config (null)
homeassistant/sensor/Solar01/AC_in_rating_voltage/config (null)
homeassistant/sensor/Solar01/AC_out_rating_active_power/config (null)
homeassistant/sensor/Solar01/AC_out_rating_apparent_power/config (null)
homeassistant/sensor/Solar01/AC_out_rating_current/config (null)
```

and now it's not. No idea why. Can't get itsending.
When I set the HA Discovery check, I start get msgs like those:
```
....
homeassistant/sensor/Solar01/PV_Charging_Power/config {"name":"PV_Charging_Power","stat_t":"Solar01/LiveData/PV_Charging_Power","avty_t":"Solar01/Alive","pl_avail": "true","pl_not_avail": "false","uniq_id":"EaSun-inv1-solar2mqtt-8C2B8E.PV_Charging_Power","ic":"mdi:solar-power-variant","unit_of_meas":"W","dev_cla":"power","dev":{"ids":["EaSun-inv1-solar2mqtt-8C2B8E"],"name":"EaSun-inv1-solar2mqtt","cu":"http://172.16.20.66","mdl":"null","mf":"SoftWareCrash","sw":"1.2.0-Pre4"}}
homeassistant/sensor/Solar01/PV_generation_day/config {"name":"PV_generation_day","stat_t":"Solar01/LiveData/PV_generation_day","avty_t":"Solar01/Alive","pl_avail": "true","pl_not_avail": "false","uniq_id":"EaSun-inv1-solar2mqtt-8C2B8E.PV_generation_day","ic":"mdi:solar-power-variant","unit_of_meas":"Wh","dev_cla":"energy","dev":{"ids":["EaSun-inv1-solar2mqtt-8C2B8E"],"name":"EaSun-inv1-solar2mqtt","cu":"http://172.16.20.66","mdl":"null","mf":"SoftWareCrash","sw":"1.2.0-Pre4"}}
homeassistant/sensor/Solar01/PV_generation_month/config {"name":"PV_generation_month","stat_t":"Solar01/LiveData/PV_generation_month","avty_t":"Solar01/Alive","pl_avail": "true","pl_not_avail": "false","uniq_id":"EaSun-inv1-solar2mqtt-8C2B8E.PV_generation_month","ic":"mdi:solar-power-variant","unit_of_meas":"Wh","dev_cla":"energy","dev":{"ids":["EaSun-inv1-solar2mqtt-8C2B8E"],"name":"EaSun-inv1-solar2mqtt","cu":"http://172.16.20.66","mdl":"null","mf":"SoftWareCrash","sw":"1.2.0-Pre4"}}
homeassistant/sensor/Solar01/PV_generation_sum/config {"name":"PV_generation_sum","stat_t":"Solar01/LiveData/PV_generation_sum","avty_t":"Solar01/Alive","pl_avail": "true","pl_not_avail": "false","uniq_id":"EaSun-inv1-solar2mqtt-8C2B8E.PV_generation_sum","ic":"mdi:solar-power-variant","unit_of_meas":"Wh","dev_cla":"energy","dev":{"ids":["EaSun-inv1-solar2mqtt-8C2B8E"],"name":"EaSun-inv1-solar2mqtt","cu":"http://172.16.20.66","mdl":"null","mf":"SoftWareCrash","sw":"1.2.0-Pre4"}}
homeassistant/sensor/Solar01/PV_generation_year/config {"name":"PV_generation_year","stat_t":"Solar01/LiveData/PV_generation_year","avty_t":"Solar01/Alive","pl_avail": "true","pl_not_avail": "false","uniq_id":"EaSun-inv1-solar2mqtt-8C2B8E.PV_generation_year","ic":"mdi:solar-power-variant","unit_of_meas":"Wh","dev_cla":"energy","dev":{"ids":["EaSun-inv1-solar2mqtt-8C2B8E"],"name":"EaSun-inv1-solar2mqtt","cu":"http://172.16.20.66","mdl":"null","mf":"SoftWareCrash","sw":"1.2.0-Pre4"}}
homeassistant/sensor/Solar01/PV_Input_Current/config {"name":"PV_Input_Current","stat_t":"Solar01/LiveData/PV_Input_Current","avty_t":"Solar01/Alive","pl_avail": "true","pl_not_avail": "false","uniq_id":"EaSun-inv1-solar2mqtt-8C2B8E.PV_Input_Current","ic":"mdi:solar-power-variant","unit_of_meas":"A","dev_cla":"current","dev":{"ids":["EaSun-inv1-solar2mqtt-8C2B8E"],"name":"EaSun-inv1-solar2mqtt","cu":"http://172.16.20.66","mdl":"null","mf":"SoftWareCrash","sw":"1.2.0-Pre4"}}
homeassistant/sensor/Solar01/PV_Input_Voltage/config {"name":"PV_Input_Voltage","stat_t":"Solar01/LiveData/PV_Input_Voltage","avty_t":"Solar01/Alive","pl_avail": "true","pl_not_avail": "false","uniq_id":"EaSun-inv1-solar2mqtt-8C2B8E.PV_Input_Voltage","ic":"mdi:solar-power-variant","unit_of_meas":"V","dev_cla":"voltage","dev":{"ids":["EaSun-inv1-solar2mqtt-8C2B8E"],"name":"EaSun-inv1-solar2mqtt","cu":"http://172.16.20.66","mdl":"null","mf":"SoftWareCrash","sw":"1.2.0-Pre4"}}
homeassistant/sensor/Solar01/Tracker_temperature/config {"name":"Tracker_temperature","stat_t":"Solar01/LiveData/Tracker_temperature","avty_t":"Solar01/Alive","pl_avail": "true","pl_not_avail": "false","uniq_id":"EaSun-inv1-solar2mqtt-8C2B8E.Tracker_temperature","ic":"mdi:thermometer-lines","unit_of_meas":"°C","dev_cla":"temperature","dev":{"ids":["EaSun-inv1-solar2mqtt-8C2B8E"],"name":"EaSun-inv1-solar2mqtt","cu":"http://172.16.20.66","mdl":"null","mf":"SoftWareCrash","sw":"1.2.0-Pre4"}}
homeassistant/sensor/Solar01/Transformer_temperature/config {"name":"Transformer_temperature","stat_t":"Solar01/LiveData/Transformer_temperature","avty_t":"Solar01/Alive","pl_avail": "true","pl_not_avail": "false","uniq_id":"EaSun-inv1-solar2mqtt-8C2B8E.Transformer_temperature","ic":"mdi:thermometer-lines","unit_of_meas":"°C","dev_cla":"temperature","dev":{"ids":["EaSun-inv1-solar2mqtt-8C2B8E"],"name":"EaSun-inv1-solar2mqtt","cu":"http://172.16.20.66","mdl":"null","mf":"SoftWareCrash","sw":"1.2.0-Pre4"}}
....
```

### Screenshots / Fotos

Wiring works fine.

### Version

1.x.x and above

### Relevant livejson output

```json
{"EspData":{"Device_name":"EaSun-inv1-solar2mqtt","ESP_VCC":3.012,"Wifi_RSSI":-41,"sw_version":"1.2.0-Pre4","Free_Heap":15552,"HEAP_Fragmentation":2,"json_memory_usage":1356,"json_capacity":2048,"runtime":22,"ws_clients":0,"detect_protocol":1},"DeviceData":{"AC_in_rating_voltage":220,"AC_in_rating_current":25.4,"AC_out_rating_voltage":220,"AC_out_rating_frequency":50,"AC_out_rating_current":25.4,"AC_out_rating_apparent_power":5600,"AC_out_rating_active_power":5600,"Battery_rating_voltage":48,"Battery_re-charge_voltage":50,"Battery_re-discharge_voltage":0,"Battery_under_voltage":46,"Battery_bulk_voltage":57,"Battery_float_voltage":57,"Battery_type":"User","Current_max_AC_charging_current":20,"Current_max_charging_current":120,"Input_voltage_range":"Appliance","Output_source_priority":"Solar-Battery-Utility","Charger_source_priority":"Solar first","Parallel_max_num":9,"Machine_type":"Off-grid Tie","Topology":"Transformerless","Output_mode":"Phase 1 of three phaseoutput","Solar_power_priority":"Load-Battery-Utility","MPPT_string":1,"Protocol_ID":18},"LiveData":{"AC_in_Voltage":240.5,"AC_in_Frequenz":50,"AC_out_Voltage":219.8,"AC_out_Frequenz":49.9,"AC_out_VA":197,"AC_out_Watt":171,"AC_out_percent":3,"Battery_Voltage":52.6,"Battery_SCC_Volt":0,"Battery_SCC2_Volt":0,"Battery_Discharge_Current":4,"Battery_Charge_Current":0,"Battery_Percent":71,"Inverter_Bus_Temperature":34,"MPPT1_Charger_Temperature":0,"MPPT2_Charger_Temperature":0,"PV1_Input_Power":0,"PV2_Input_Power":0,"PV1_Input_Voltage":0,"PV2_Input_Voltage":0,"Configuration_State":0,"MPPT1_Charger_Status":0,"MPPT2_CHarger_Status":0,"Load_Connection":1,"Battery_Power_Direction":2,"ACDC_Power_Direction":2,"Line_Power_Direction":0,"Local_Parallel_ID":0,"PV_Input_Voltage":0,"PV_Charging_Power":0,"PV_Input_Current":0,"Battery_Load":-4,"Inverter_Operation_Mode":"Battery","Time_until_absorb_charge":0,"Time_until_float_charge":10,"Tracker_temperature":0,"Inverter_temperature":0,"Battery_temperature":0,"Transformer_temperature":34,"Fan_speed":0,"Inverter_charge_state":30,"PV_generation_day":0,"PV_generation_month":55599,"PV_generation_year":55599,"PV_generation_sum":55599}}
```

### What browsers are you seeing the problem on?

Chrome

valqk, 2024-10-17:
BTW, this option is incorrect:
"Output_mode": "Phase 1 of three phaseoutput",
The output mode is in parallel (currently 2 inverters in paralel, having issues wtih 3rd one - says it's in Phase 3 of three phaseout and rises error when turned on. Still debuging). This one is working fine with the other one though.

softwarecrash, 2024-10-18:
First, please try the Last non-pre Release, the pre releases are Beta and can have Bugs.

valqk, 2024-10-18:
OK, I'll try it, but how to build the project? Are you using vscode? The web installer has only the pre release avaliable to install...

softwarecrash, 2024-10-18:
you dont need to compile itself, use ready to use files
https://github.com/softwarecrash/Solar2MQTT/releases
use tasmotizer to flash it or over webui firmware update

valqk, 2024-10-18:
Hm, just started to try to adopt it to Arduino IDE and came up with #include <ESPAsyncWiFiManager.h> is missing. Lib installed but include requires #include <ESPAsync_WiFiManager.h>
I'll try with the binary. How do you compile it anyway?

valqk, 2024-10-18:
Thanks. Will try vscode+pio.
In 1.1.8 values are populated to MQTT properly and HA can see them.
Should I close that bug or leave it open, so I test the Pre4/5 when you say it's fixed?

softwarecrash, 2024-10-18:
can you check with mqttexplorer that what values are send? so i can fix it, leave it open for future testing.
for the output mode, can you copy me the debug output from mqtt, its under /raw

valqk, 2024-10-18:
"can you check with mqttexplorer that what values are send? so i can fix it, leave it open for future testing."
You mean in 1.2.0-Pre4? Should I put it back and paste you the raw mqtt string?

-------
This are all the values I now get (working with 1.1.8). It's raw text from the mqtt (I have no mqttexplorer, or at least I'm not finding one, neither in ha nor in esp web).
Is this OK for the output mode?

```
Solar01/EspData/Device_name EaSun-inv1-solar2mqtt
Solar01/EspData/ESP_VCC 3.014
Solar01/EspData/Wifi_RSSI -47
Solar01/EspData/sw_version 1.1.8
Solar01/EspData/Free_Heap 23832
Solar01/EspData/HEAP_Fragmentation 22
Solar01/EspData/json_memory_usage 1519
Solar01/EspData/json_capacity 2048
Solar01/EspData/runtime 11144
Solar01/EspData/ws_clients 0
Solar01/EspData/detect_protocol 1
Solar01/DeviceData/AC_in_rating_voltage 220
Solar01/DeviceData/AC_in_rating_current 25.4
Solar01/DeviceData/AC_out_rating_voltage 220
Solar01/DeviceData/AC_out_rating_frequency 50
Solar01/DeviceData/AC_out_rating_current 25.4
Solar01/DeviceData/AC_out_rating_apparent_power 5600
Solar01/DeviceData/AC_out_rating_active_power 5600
Solar01/DeviceData/Battery_rating_voltage 48
Solar01/DeviceData/Battery_re-charge_voltage 50
Solar01/DeviceData/Battery_re-discharge_voltage 0
Solar01/DeviceData/Battery_under_voltage 46
Solar01/DeviceData/Battery_bulk_voltage 57
Solar01/DeviceData/Battery_float_voltage 57
Solar01/DeviceData/Battery_type User
Solar01/DeviceData/Current_max_AC_charging_current 20
Solar01/DeviceData/Current_max_charging_current 120
Solar01/DeviceData/Input_voltage_range Appliance
Solar01/DeviceData/Output_source_priority Solar-Battery-Utility
Solar01/DeviceData/Charger_source_priority Solar first
Solar01/DeviceData/Parallel_max_num 9
Solar01/DeviceData/Machine_type Off-grid Tie
Solar01/DeviceData/Topology Transformerless
Solar01/DeviceData/Output_mode Phase 1 of three phaseoutput
Solar01/DeviceData/Solar_power_priority Load-Battery-Utility
Solar01/DeviceData/MPPT_string 1
Solar01/DeviceData/Protocol_ID 18
Solar01/LiveData/AC_in_Voltage 232.8
Solar01/LiveData/AC_in_Frequenz 49.9
Solar01/LiveData/AC_out_Voltage 220
Solar01/LiveData/AC_out_Frequenz 50
Solar01/LiveData/AC_out_VA 1675
Solar01/LiveData/AC_out_Watt 1675
Solar01/LiveData/AC_out_percent 29
Solar01/LiveData/Battery_Voltage 52.1
Solar01/LiveData/Battery_SCC_Volt 0
Solar01/LiveData/Battery_SCC2_Volt 0
Solar01/LiveData/Battery_Discharge_Current 35
Solar01/LiveData/Battery_Charge_Current 0
Solar01/LiveData/Battery_Percent 77
Solar01/LiveData/Inverter_Bus_Temperature 42
Solar01/LiveData/MPPT1_Charger_Temperature 0
Solar01/LiveData/MPPT2_Charger_Temperature 0
Solar01/LiveData/PV1_Input_Power 0
Solar01/LiveData/PV2_Input_Power 0
Solar01/LiveData/PV1_Input_Voltage 0
Solar01/LiveData/PV2_Input_Voltage 0
Solar01/LiveData/Configuration_State 0
Solar01/LiveData/MPPT1_Charger_Status 0
Solar01/LiveData/MPPT2_CHarger_Status 0
Solar01/LiveData/Load_Connection 1
Solar01/LiveData/Battery_Power_Direction 2
Solar01/LiveData/ACDC_Power_Direction 2
Solar01/LiveData/Line_Power_Direction 0
Solar01/LiveData/Local_Parallel_ID 0
Solar01/LiveData/PV_Input_Voltage 0
Solar01/LiveData/PV_Charging_Power 0
Solar01/LiveData/PV_Input_Current 0
Solar01/LiveData/Battery_Load -35
Solar01/LiveData/Inverter_Operation_Mode Battery
Solar01/LiveData/Time_until_absorb_charge 0
Solar01/LiveData/Time_until_float_charge 26
Solar01/LiveData/Tracker_temperature 0
Solar01/LiveData/Inverter_temperature 0
Solar01/LiveData/Battery_temperature 0
Solar01/LiveData/Transformer_temperature 42
Solar01/LiveData/Fan_speed 0
Solar01/LiveData/Inverter_charge_state 30
Solar01/LiveData/PV_generation_day 6343
Solar01/LiveData/PV_generation_month 61942
Solar01/LiveData/PV_generation_year 61942
Solar01/LiveData/PV_generation_sum 2300
Solar01/RAW/Q1 0 00026 00 00 00 000 042 030 040 02 00 000 0030 0000 0000 49.95 10 0 000 000 000 000 00.00 000 000 0 0000
Solar01/RAW/QPIGS 2328,499,2200,500,1675,1675,029,521,000,000,035,000,077,042,000,000,0000,0000,0000,0000,0,0,0,1,2,2,0,0
Solar01/RAW/QPIGS2 (null)
Solar01/RAW/QPIRI 2200,254,2200,500,254,5600,5600,480,500,000,460,570,570,2,020,120,0,1,0,9,0,0,1,1,1,00
Solar01/RAW/QT 20241018193003
Solar01/RAW/QET 00002300
Solar01/RAW/QEY 00061942
Solar01/RAW/QEM 00061942
Solar01/RAW/QED 00006343
Solar01/RAW/QLT 20241018193003
Solar01/RAW/QLY (null)
Solar01/RAW/QLM (null)
Solar01/RAW/QLD (null)
Solar01/RAW/QPI 18
Solar01/RAW/QMOD 03
Solar01/RAW/QALL (null)
Solar01/RAW/QMN (null)
Solar01/Alive true
Solar01/EspData/Wifi_RSSI -45

```

softwarecrash, 2024-10-18:
i think i found the parallel mode issue
try this

[Solar2MQTT_d1_mini_1.2.0-Pre4A1.zip](https://github.com/user-attachments/files/17438605/Solar2MQTT_d1_mini_1.2.0-Pre4A1.zip)

valqk, 2024-10-18:
I saw the raw 1675 values, and suspected it was wats/va. Interestingly the inverter itself shows correct difference on display and in SolarPower app. Maybe they do calculations, but for those, power factor is needed, and I don't see it here... 

Does A2 includes A1 fixes or I should try them separately?
About the temps and fan:
    Tracker_temperature -> Shows zero, which is not correct. (check above)
    Inverter_temperature = 0 - incorrect
    Battery_temperature = 0 - not correct
    Transformer_temperature = 42 - which is about right (I suspect internal temp - so maybe it's correct)

softwarecrash, 2024-10-18:
A2 includes A1
So this values inkl Fan Speed are Data from the Q1 command, this IS a undocumented debug command. Some devices gives garbage on this command.
Yes, in Display it Shows correct values and with the known querry commands it gives wrong values, that are Things that i cant change

valqk, 2024-10-18:
Understandable (for the undocumented commands and wrong values). Just reporting and saying if there's power factor in the packet avaliable the va can be calculated from w... not that important to me anyway!
That's interesting - in livejson, va and w looks correct now:
   "AC_out_VA": 242,
    "AC_out_Watt": 203,
Can't check in HA because of the unkown/null bug.

Just installed the A2 bin and:
- now fan_speed shows 30 (not sure if thats % or something but consumption is low now and I suspect it's correct. I'll test tommorow, when batteries are charged and fans are spinning faster if value changes accordingly).
- output mode says: parallel output (the following days I'll have to debug what's wrong with the third inverter and will write here how param reacts upon setting itfrom SolarPower app to other values).
- HA got all params to unkown again. (maybe that's not fixed yet)

softwarecrash, 2024-10-19:
now installed HA and tested it, with my device emulator (pi30) it works in HA without any problems.
you can use http://mqtt-explorer.com/ and connect to your mqtt broker, so you have a overview of all data.
the esp push the data what you allready listed like `Solar01/Alive true` and more, so with HA discovery it pushes for every state a config for HA
[image]
so it can help to delete the device in HA and restart the ESP, badly HA made some rentained entrys, need some manual adjustment

valqk, 2024-10-19:
Yes, I know what it pushes - I connect to mosquito mqtt from the cli. I see all the config options, but when it's on 1.1.8
I see this:
```
Solar01/LiveData/Time_until_absorb_charge 0
Solar01/LiveData/Time_until_float_charge 83
Solar01/LiveData/Tracker_temperature 0
Solar01/LiveData/Inverter_temperature 0
Solar01/LiveData/Battery_temperature 0
Solar01/LiveData/Transformer_temperature 42
Solar01/LiveData/Fan_speed 0
Solar01/LiveData/Inverter_charge_state 97
Solar01/LiveData/PV_generation_day 8415
Solar01/LiveData/PV_generation_month 70333
Solar01/LiveData/PV_generation_year 70318
Solar01/LiveData/PV_generation_sum 70321
Solar01/RAW/Q1 0 00083 00 00 00 000 042 034 042 02 00 000 0097 0000 0000 49.98 12 0 000 000 000 000 00.00 000 000 0 0000
Solar01/RAW/QPIGS 2344,501,2191,499,0350,0190,006,550,000,000,000,055,082,043,000,000,3504,0000,3632,0000,0,2,0,1,1,2,0,0
Solar01/RAW/QPIGS2 (null)
Solar01/RAW/QPIRI 2200,254,2200,500,254,5600,5600,480,500,000,460,570,570,2,020,120,0,1,0,9,0,0,1,1,1,00
Solar01/RAW/QT 20241019125535
Solar01/RAW/QET 00070321
Solar01/RAW/QEY 00070318
Solar01/RAW/QEM 00070333
Solar01/RAW/QED 00008415
Solar01/RAW/QLT 20241019125535
Solar01/RAW/QLY (null)
Solar01/RAW/QLM (null)
Solar01/RAW/QLD (null)
Solar01/RAW/QPI 18
Solar01/RAW/QMOD 03
Solar01/RAW/QALL (null)
Solar01/RAW/QMN (null)
Solar01/Alive true
Solar01/EspData/Wifi_RSSI -46

```

I have just put the A2 fix from yesterday to test the fan speed (now it's charging).
The only strings with 1.2.0-PRE4A2 in mqtt (beside all the config strings) I get are:

```
Solar01/Alive false
Solar01/Alive true
Solar01/IP 172.16.20.66
```
ESP will send config values periodically - strings like 
`homeassistant/sensor/Solar01/Transformer_temperature/config {"name":"Transformer_temperature","stat_t":"Solar01/LiveData/Transformer_temperature","avty_t":"Solar01/Alive","pl_avail": "true","pl_not_avail": "false","uniq_id":"EaSun-inv1-solar2mqtt-8C2B8E.Transformer_temperature","ic":"mdi:thermometer-lines","unit_of_meas":"°C","dev_cla":"temperature","dev":{"ids":["EaSun-inv1-solar2mqtt-8C2B8E"],"name":"EaSun-inv1-solar2mqtt","cu":"http://172.16.20.66","mdl":"null","mf":"SoftWareCrash","sw":"1.2.0-Pre4A2"}}
`
but no 
`Solar01/...live value...
`

and nothing else, thus in HA all values becomes unavailable. This strings are sent initially after reboot and after that no values are sent. Any ideas why esp won't send any values in mqtt and have them in livejson?

About the fan speed. Now the inverter is charging batteries and fan speed it at 45. When Idle (below 1kw consumption) it's on 30. I'll have to compare the value with the one in SolarPower app but looks like working.

softwarecrash, 2024-10-19:
> I connect to mosquito mqtt from the cli. I see all the config options, but when it's on 1.1.8 I see this:
> 
> ```
> Solar01/LiveData/Time_until_absorb_charge 0
> Solar01/LiveData/Time_until_float_charge 83
> Solar01/LiveData/Tracker_temperature 0
> Solar01/LiveData/Inverter_temperature 0
> Solar01/LiveData/Battery_temperature 0
> Solar01/LiveData/Transformer_temperature 42
> Solar01/LiveData/Fan_speed 0
> Solar01/LiveData/Inverter_charge_state 97
> Solar01/LiveData/PV_generation_day 8415
> Solar01/LiveData/PV_generation_month 70333
> Solar01/LiveData/PV_generation_year 70318
> Solar01/LiveData/PV_generation_sum 70321
> Solar01/RAW/Q1 0 00083 00 00 00 000 042 034 042 02 00 000 0097 0000 0000 49.98 12 0 000 000 000 000 00.00 000 000 0 0000
> Solar01/RAW/QPIGS 2344,501,2191,499,0350,0190,006,550,000,000,000,055,082,043,000,000,3504,0000,3632,0000,0,2,0,1,1,2,0,0
> Solar01/RAW/QPIGS2 (null)
> Solar01/RAW/QPIRI 2200,254,2200,500,254,5600,5600,480,500,000,460,570,570,2,020,120,0,1,0,9,0,0,1,1,1,00
> Solar01/RAW/QT 20241019125535
> Solar01/RAW/QET 00070321
> Solar01/RAW/QEY 00070318
> Solar01/RAW/QEM 00070333
> Solar01/RAW/QED 00008415
> Solar01/RAW/QLT 20241019125535
> Solar01/RAW/QLY (null)
> Solar01/RAW/QLM (null)
> Solar01/RAW/QLD (null)
> Solar01/RAW/QPI 18
> Solar01/RAW/QMOD 03
> Solar01/RAW/QALL (null)
> Solar01/RAW/QMN (null)
> Solar01/Alive true
> Solar01/EspData/Wifi_RSSI -46
> ```
> 
> I have just put the A2 fix from yesterday to test the fan speed (now it's charging). The only strings with 1.2.0-PRE4A2 in mqtt (beside all the config strings) I get are:
> 
> ```
> Solar01/Alive false
> Solar01/Alive true
> Solar01/IP 172.16.20.66
> ```
> 
> ESP will send config values periodically - strings like `homeassistant/sensor/Solar01/Transformer_temperature/config {"name":"Transformer_temperature","stat_t":"Solar01/LiveData/Transformer_temperature","avty_t":"Solar01/Alive","pl_avail": "true","pl_not_avail": "false","uniq_id":"EaSun-inv1-solar2mqtt-8C2B8E.Transformer_temperature","ic":"mdi:thermometer-lines","unit_of_meas":"°C","dev_cla":"temperature","dev":{"ids":["EaSun-inv1-solar2mqtt-8C2B8E"],"name":"EaSun-inv1-solar2mqtt","cu":"http://172.16.20.66","mdl":"null","mf":"SoftWareCrash","sw":"1.2.0-Pre4A2"}} ` but no `Solar01/...live value... `
> 
> and nothing else, thus in HA all values becomes unavailable. Tried deleting/adding inverter device in HA with no luck - but if it won't get Solar01 live values in mqtt - that's expected to not show them in HA. This strings are sent initially after reboot and after that no values are sent. Any ideas why esp won't send any values in mqtt and have them in livejson?
> 
> About the fan speed. Now the inverter is charging batteries and fan speed it at 45. When Idle (below 1kw consumption) it's on 30. I'll have to compare the value with the one in SolarPower app but looks like working. I'll get back about the fan speed when I compare the value shown in SolarPower compared to the esp value so we can adjust the speed to % but not to bit value.

okay, periodicly is not normal, thats sounds the esp crashes and restart, let me rebuild my emulator with your P18 protocol type and check whats going wrong, need a bit of time

valqk, 2024-10-19:
If you need some dumps from the raw communication - let me know. I can connect serial and check if there are any trace dump that can point to more precise location that's crashing.
Maybe it'd be easier if we connect in some chat or something for easier communication?

softwarecrash, 2024-10-19:
RAW dumps from serial are helpfull.
You can Join the discord Channel, linked in the top of the readme

closed 2024-12-03 by github-actions[bot]