softwarecrash/Solar2MQTT-ESP8266 #159: [FEATURE] Multiple inverters on single ESP?

valqk, 2024-10-17:
### Is your feature request related to a problem? Please describe

I have 3 inverters in parallel. I use Nocemcu v3. I'd like to be able to connect all 3 inverters to 1 esp and have all 3 to be reported with only 1 esp.

### Describe the solution you'd like

If >1 inverter is connected to a single ESP this means there should be options to select pins<->inverter + topic/name for each inverter.

### Describe alternatives you've considered

If that's not possible I will install 3 esps, but I see no point when we have so much free pins.
Another alternative is to test with esp01, but I suspect it will be a pain to get it running because of the limited pin count and no option for debugging.

### Additional context

I'm not sure if the cpu will be able to handle >1 communication + if you will want to over-complicate the config, etc.
Just wanted to ask what do you think about such an option?
Great work! Thank you very much!

softwarecrash, 2024-10-18:
Hello,
Actual the Firmware isnt desiged for Support multiple devices at one ESP.
I think its Not possible, the ESP have Not enugh RAM for Holding all Data for multiple devices.

valqk, 2024-10-18:
I'll make some tests with the esp01. I was thinking that there might be not enough resources, just wanted to share idea.
Thank you.

closed 2024-10-18 by valqk