softwarecrash/Solar2MQTT-ESP8266 #160: [Bug]: Wrong solar current; wrong solar power

ichbins0815, 2024-10-18:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

I'm using Solar2MQTT with an Effekta 5000 AX M1.
It works fine with only one issue.
The PV_Input_Current is the same as Battery_Load.
The PV_Input_Current is wrong, the Battery_Load looks right.
Also the PV_Input_Power is wrong.
Same data in the webinterface and  via MQTT.

I've tried the Version 1.2.1(PRE), same wrong data.

### Screenshots / Fotos

[image]
[image]

### Steps To Reproduce

Look at webinterface for the same current (Solar and Battery)

### Version

1.x.x and above

### Relevant livejson output

```json
EspData	
Device_name	"Solar2MQTT"
ESP_VCC	3.026
Wifi_RSSI	-55
sw_version	"1.1.8"
Free_Heap	21688
HEAP_Fragmentation	20
json_memory_usage	1201
json_capacity	2048
runtime	329
ws_clients	1
detect_protocol	2
DeviceData	
AC_in_rating_voltage	230
AC_in_rating_current	21.7
AC_out_rating_voltage	230
AC_out_rating_frequency	50
AC_out_rating_current	21.7
AC_out_rating_apparent_power	5000
AC_out_rating_active_power	5000
Battery_rating_voltage	48
Battery_re-charge_voltage	49
Battery_under_voltage	48
Battery_bulk_voltage	55.8
Battery_float_voltage	55
Battery_type	"User"
Current_max_AC_charging_current	30
Current_max_charging_current	70
Input_voltage_range	"Appliance"
Output_source_priority	"SBU first"
Charger_source_priority	"Only solar charging permitted"
Parallel_max_num	9
Machine_type	"Off Grid"
Topology	"Transformerless"
Output_mode	0
Battery_re-discharge_voltage	54
PV_OK_condition_for_parallel	0
PV_power_balance	1
Max_charging_time_at_CV_stage	0
Protocol_ID	"PI30"
LiveData	
AC_in_Voltage	0
AC_in_Frequenz	0
AC_out_Voltage	230.1
AC_out_Frequenz	50
AC_out_VA	23
AC_out_Watt	0
AC_out_percent	0
Inverter_Bus_Voltage	424
Battery_Voltage	52.8
Battery_Charge_Current	4
Battery_Percent	46
Inverter_Bus_Temperature	35
PV_Input_Current	4
PV_Input_Voltage	97.4
Battery_SCC_Volt	52.83
Battery_Discharge_Current	0
Status_Flag	110110
Battery_voltage_offset_fans_on	0
EEPROM_Version	0
PV_Charging_Power	232
Device_Status	10
Battery_Load	4
PV_Input_Power	388
Inverter_Operation_Mode	"Battery"
```

### What browsers are you seeing the problem on?

Firefox

softwarecrash, 2024-10-19:
Can you share the RAW Data from mqtt?

ichbins0815, 2024-10-19:
Hi softwarecrash,
I hope this are the right one ;-)

RAW
Q1 = 00000 00000 01 01 00 033 024 028 033 00 00 000 0030 0338 0000 00.00 11 0 060 000 070 000 55.00 000 120 0 0000
QPIGS = 000.0 00.0 230.1 50.0 0023 0000 000 419 52.20 006 040 0033 0006 097.1 52.12 00000 00110110 00 00 00333 010
QPIGS2 = NAK
QPIRI = 230.0 21.7 230.0 50.0 21.7 5000 5000 48.0 49.0 48.0 55.8 55.0 2 30 070 0 2 3 9 01 0 0 54.0 0 1 000
QT = {}
QET = NAK
QEY = {}
QEM = {}
QED = {}
QLT = {}
QLY = {}
QLM = {}
QLD = {}
QPI = PI30
QMOD = B
QALL = NAK
QMN = NAK

softwarecrash, 2024-10-19:
your device doesnt have this value, so it send "pv input current for battery" this is the same as battery current

ichbins0815, 2024-10-19:
Not realy.
In the software (Watchpower) I can see PV current and PV Power.
A lot of years I used the Software "Solaranzeige " and there I can
also see the data.
Regards

softwarecrash, 2024-10-19:
then enable debug log on watchpower (right click on the smal icon in task and enable debug) there produces log files, in this log files you must find your pv current value.
in your raw data actual it shows QPIGS
take a look at the datasheet https://github.com/softwarecrash/Solar2MQTT/blob/master/Protocol/PI30_HS_MS_MSX_RS232_Protocol_20140822_after_current_upgrade.pdf
under QPIGS, i cant find PV current.
possible that watchpower or solaranzeige calculate this value?

ichbins0815, 2024-10-19:
Sehr gerne auf Deutsch, dachte nur das anderen Lesern vieleicht "schlechtes" Englisch hilft.

Ich habe bis zum Tod meiner SD-Karte im RasPi "Solaranzeige" benutzt.
War aber schon länger auf der Suche nach einer Option für Home Assistant.
Habe eine fertige Platine bestellt, Software eingerichtet und - läuft. Klasse.
Habe aber nur das Kabel umgesteckt. Die passenden Werte sollte der
Wechselrichter also ausspucken.
Ich meine mich zu erinnern den gleichen Fehler mit dem Entwickler von "Solaranzeige"
diskutiert zu haben. Der hat dann auch ein wenig suchen müssen.
Bin nicht so firm im Programmieren, wie kann ich helfen?

Ich werde Watchpower mal anwerfen und in den logs suchen.
Gruß

ichbins0815, 2024-10-19:
PS.
Vieleicht hast Du recht mit der Berechnung des Wertes für den PV-Strom.
Ich muß klären ob PV-Leistung > Batterie-Leistung ist.
Gruß

ichbins0815, 2024-10-20:
Hallo Leute,
so auf die Schnelle aus "Watchpower":
Es gibt Werte für PV-Leistung und PV-Spannung.
Aber die Verbindung via USB. Aber warum sollten
die Werte via RS232 andere sein.
Gruß
[image]

softwarecrash, 2024-10-20:
> Hallo Leute, so auf die Schnelle aus "Watchpower": Es gibt Werte für PV-Leistung und PV-Spannung. Aber die Verbindung via USB. Aber warum sollten die Werte via RS232 andere sein. Gruß ![Watchpower1](https://private-user-images.githubusercontent.com/184989024/378195718-5dddafcc-95e1-41a4-9148-b79a55f4551a.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk0NDgzNjcsIm5iZiI6MTcyOTQ0ODA2NywicGF0aCI6Ii8xODQ5ODkwMjQvMzc4MTk1NzE4LTVkZGRhZmNjLTk1ZTEtNDFhNC05MTQ4LWI3OWE1NWY0NTUxYS5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAyMFQxODE0MjdaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xMDQ0NDFjZjM1NzVmOWZhZDkwMzhlNGQ2Y2Q1YzNmNmIwZmY2MjEzZjk4NjM2ZDI5N2ViM2M4MGM4NzQzNWVjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.sDigO-85E_cWHFYhrAA--ODvHpkp9_JL9QmPRrjrbMs)

Ich sehe da keine PV Ampere, nur Watt, und die werden auch übertragen.

ichbins0815, 2024-10-21:
@softwarecrash
eben, es wird kein PV Strom angezeigt, der muss wohl berechnet werden.
Im solar2mppt wird er angezeigt.
@all-solutions
ich habe kein debug in watchpower gefunden. Leider nur historydatas.
Nutze V1.11
[image]

softwarecrash, 2024-10-22:
> Soooo, neuere Watchpower-Version und schon gibt es ein LOG ;-) Auf jeden Fall schon mal vielen Dank für Eure Mühe und die viele Arbeit in diesem Projekt. Gruß
> 
> [debug.zip](https://github.com/user-attachments/files/17473186/debug.zip)

jetzt müsstest du das log nur in einem zeitraum anlegen wo auch pv current angezeigt und vorhanden ist und uns dann sagen nach welchem wert wir suchen müssen

ichbins0815, 2024-10-22:
So nun etwas aufbereitet:
Wie gesagt es gibt anscheinend keine Werte für PV Strom und Batterieleistung.
Die log's kommen aus Watchpower V1.17SP4 via USB.
Die Werte sind von 11:43 mit Lasst und von 11:44 ohne.
Die Werte in rot sind errechnet. Ich habe sie nirgends gefunden.
[1143-1144QPIGS.pdf](https://github.com/user-attachments/files/17474966/1143-1144QPIGS.pdf)
[2024-10-22 USB-QPGS0.log](https://github.com/user-attachments/files/17474967/2024-10-22.USB-QPGS0.log)
[2024-10-22 USB-QPIGS.log](https://github.com/user-attachments/files/17474968/2024-10-22.USB-QPIGS.log)
Ich hoffe das hilft, danke

softwarecrash, 2024-10-22:
okay, also wie schon gedacht gibt es keine echten daten für current, das wird errechnet.
es gibt lediglich pv current for battery charge, aber das ist am ende das was von pv in die batterie fliest ohne verluste oder abzweigung ins ac

ichbins0815, 2024-10-22:
"...das wird errechnet" von wem?

"...pv current for battery charge, aber das ist am ende das was von pv in die batterie fliest..." macht ja keinen Sinn.
Der Strom in die Batterie ist ja um einiges größer als der PV Strom.

ichbins0815, 2024-10-22:
NACHGEREICHT
[image]
[image]

ichbins0815, 2024-10-22:
Wenn das alle Werte sind die der Regler ausspuckt, müßte man folgendes berechnen:
"PV_Charging_Power" ist eigendlich "PV_Input_Power"
"PV_Input_Current" wird berechnet aus "PV_Input_Power" / "PV_Input_Voltage"
"PV_Charging_Power" wird berechnet aus "Battery_Voltage" X "Battery_Load"
Wobei "PV_Charging_Power" besser "Battery_Charging_Power" heißen sollte.
Gruß

softwarecrash, 2024-10-22:
> Wenn das alle Werte sind die der Regler ausspuckt, müßte man folgendes berechnen: "PV_Charging_Power" ist eigendlich "PV_Input_Power" "PV_Input_Current" wird berechnet aus "PV_Input_Power" / [...]
ich habe die benennung großteils so aus den datenblättern übernommen
https://github.com/softwarecrash/Solar2MQTT/blob/master/Protocol/PI30MAX.Communication.Protocol20210217.pdf

> "PV_Charging_Power" ist eigendlich "PV_Input_Power"
falsch, es ist die ladeleistung, wenn dein inverter zur batterie und gleichzeitig in ac speisen kann ist das der wert der in die batterie geht.

ichbins0815, 2024-10-22:
"falsch, es ist die ladeleistung, wenn dein inverter zur batterie und gleichzeitig in ac speisen kann ist das der wert der in die batterie geht."
Sorry das stimmt nicht! Siehe unten. Es wird auf Grund der Last von ca.1,85kW Strom aus der Batterie entnommen.
Also eine Ladeleistung von ca. -1726W; der Rest kommt direkt von der Sonne.
[image]

ichbins0815, 2024-10-22:
Die Werte in HA zeigen das gleiche

softwarecrash, 2024-10-22:
> Die Werte in HA zeigen das gleiche

na das die werte der webUI und HA / MQTT identisch sind sollte doch wohl klar sein?

es werden die daten geliefert die der inverter laut protokol liefert, lediglich ein zwei watt werte werden zusätzlich grob anhand vorhandener werte berechnet, einfach aus nettigkeit, aber wenn die falsch sind. igonrier sie, real ist das was du in dem protokol und dem log findest, inwieweit das korrekt ist und stimm weis nur dein inverter.

ichbins0815, 2024-10-22:
OK, das erschlägt mich erstmal. Aber das war ja Deine Absicht, oder ;-)
Ich werde mich mal durch arbeiten und vieleicht kann ich mir und anderen helfen.
Denn das war meine Absicht.
Es steht mir nicht zu Dir zu sagen was Du tun sollst. Ich freue mich über
die Menschen die sich bei so etwas so viel Mühe geben und anderen das Ergebnis
zur Verfügung stellen.
Aber bleiben wir sachlich.
Dein blau markierter Wert oben ist doch "Battery_Charge_Current //KKK" oder?
Was ja stimmen würde.

Wie gesagt meine Werte kommen über die USB Schnittstelle. Sollte ja identisch mit der RS232 sein.

softwarecrash, 2024-10-22:
die daten über usb oder serial sind absulut identisch (im usb ist nur ein rs wandler)
gern kannst du schauen ob die zuordnungen korrekt sind und ggf neu sortieren, dann brauche ich aber aus dem webserial noch von der zeile wo qpigs auftaucht die werte um es zuzuordnen, und ggf das genaue gerätemodel

closed 2024-12-06 by github-actions[bot]