softwarecrash/Solar2MQTT-ESP8266 #162: [HELP] EASUN SMH II 4.2KW DONT WORK

avenseElectro, 2024-10-19:
### A clear and concise description of what the problem is.

I can't get this inverter to work at all, can anyone help me?

```

[WebSerial] Connecting...
[WebSerial] Connected...
[C: ^P005PI][CR:    0][CC:    0][L:   3]

^P005PI:		NAK (Length: 3)

Try Autodetect Modbus device

Invalid Slave ID

Invalid Slave ID

Time-out

Request failed!

Invalid Slave ID

Invalid Slave ID

Time-out

Request failed!

Invalid Slave ID

Invalid Slave ID

Time-out

Request failed!

Invalid Slave ID

Invalid Slave ID

Time-out

Request failed!
----------------- End Autodetect -----------------
----------------- Start Autodetect -----------------

[C:   QPI][CR:    0][CC:    0][L:   3]

QPI:		NAK (Length: 3)

[C: ^P005PI][CR:    0][CC:    0][L:   3]

^P005PI:		NAK (Length: 3)

Try Autodetect Protocol

[C:   QPI][CR:    0][CC:    0][L:   3]

QPI:		NAK (Length: 3)

[C: ^P005PI][CR:    0][CC:    0][L:   3]

^P005PI:		NAK (Length: 3)

Try Autodetect Protocol

[C:   QPI][CR:    0][CC:    0][L:   3]

QPI:		NAK (Length: 3)

[C: ^P005PI][CR:    0][CC:    0][L:   3]

^P005PI:		NAK (Length: 3)

Try Autodetect Modbus device

Invalid Slave ID

Invalid Slave ID

Time-out

Request failed!

Invalid Slave ID

Invalid Slave ID

Time-out

Request failed!
Invalid Slave ID

Invalid Slave ID

Time-out

Request failed!
Invalid Slave ID

Invalid Slave ID

Time-out

Request failed!
----------------- End Autodetect -----------------
```

softwarecrash, 2024-10-19:
the smh series works so i think, can you explain your wireing, photos and more

avenseElectro, 2024-10-19:
> the smh series works so i think, can you explain your wireing, photos and more

The connection is made as you see in your diagram...
On pipsolar it works well and I don't have any problems. I switch to SMH and I can't get it to work

softwarecrash, 2024-10-19:
> > the smh series works so i think, can you explain your wireing, photos and more
> 
> The connection is made as you see in your diagram... On pipsolar it works well and I don't have any problems. I switch to SMH and I can't get it to work

pipsolar have a lot of implemented protocols, i dont have all at the moment.
without know what your device speaks i cant fix it.
double check your wireing for twisted data lines and/or try to swap it

avenseElectro, 2024-10-20:
> pipsolar have a lot of implemented protocols, i dont have all at the moment. without know what your device speaks i cant fix it. double check your wireing for twisted data lines and/or try to swap it

I just find it strange that the pipsolar I have (pip30) works fine and when I change to the Easun smh II 4.2kw it stops working.
But ok I'll check the connections but I'm almost certain that it's clear that the pipsolar where it uses the pip30 protocol works fine.

softwarecrash, 2024-10-20:
Pipsolar have a Lot more as the single pi30 Protocol, WE need to find Out what your device Talk in Details.
Can you get a RAW or debug Log of communication from your device?
Or use watchppwer to created a communication log

avenseElectro, 2024-10-21:
I've tried several times and re-read the connections and they're all ok.
Remembering that just connect to the pipsolar inverter (pip30 protocol works)
Tell me how to debug watchpower....

softwarecrash, 2024-10-21:
> I've tried several times and re-read the connections and they're all ok. Remembering that just connect to the pipsolar inverter (pip30 protocol works) Tell me how to debug watchpower....

start watchpower, right click on the litle icon on the task bar(next to clock) and click on debug, a folder opens. when watchpower is connected and read data the folder will fill up with files. zip it and upload it here

avenseElectro, 2024-10-21:
> start watchpower, right click on the litle icon on the task bar(next to clock) and click on debug, a folder opens. when watchpower is connected and read data the folder will fill up with files. zip [...]

What do you need from here?

[image]

avenseElectro, 2024-10-21:
> start watchpower, right click on the litle icon on the task bar(next to clock) and click on debug, a folder opens. when watchpower is connected and read data the folder will fill up with files. zip [...]

Here is the watchpower debug.

[debug.zip](https://github.com/user-attachments/files/17464959/debug.zip)

softwarecrash, 2024-10-21:
thank you, so the little problem is simple.
i ask the devices for QPI the most answer with 18 or 30 was the protocol version is, but yours doenst answer with anything.
short crude fix, but try it
[Solar2MQTT_1.2.0-Pre5A3_OTA.bin.zip](https://github.com/user-attachments/files/17465522/Solar2MQTT_1.2.0-Pre5A3_OTA.bin.zip)

avenseElectro, 2024-10-21:
> thank you, so the little problem is simple. i ask the devices for QPI the most answer with 18 or 30 was the protocol version is, but yours doenst answer with anything. short crude fix, but try it [...]

Unfortunately it also doesn't work with the version you sent.

softwarecrash, 2024-10-21:
> > thank you, so the little problem is simple. i ask the devices for QPI the most answer with 18 or 30 was the protocol version is, but yours doenst answer with anything. short crude fix, but try it [Solar2MQTT_1.2.0-Pre5A3_OTA.bin.zip](https://github.com/user-attachments/files/17465522/Solar2MQTT_1.2.0-Pre5A3_OTA.bin.zip)
> 
> Unfortunately it also doesn't work with the version you sent.
> 
> But you have it on your github stating that this inverter works.... Why isn't mine working?

possible that your device have different firmware
what tells the web serial now?

avenseElectro, 2024-10-21:
> possible that your device have different firmware what tells the web serial now?

```
[WebSerial] Connecting...
[WebSerial] Connected...
[C:   QPI][CR:    0][CC:    0][L:   3]

QPI:		NAK (Length: 3)

[C:  QVFW][CR:    0][CC:    0][L:   3]

QVFW:		NAK (Length: 3)

[C: ^P005PI][CR:    0][CC:    0][L:   3]

^P005PI:		NAK (Length: 3)

Try Autodetect Modbus device

Invalid Slave ID

Invalid Slave ID

Time-out
Invalid Slave ID

Invalid Slave ID

Invalid Slave ID

Invalid Slave ID

Time-out

Invalid Slave ID

Invalid Slave ID

Time-out

----------------- Start Autodetect -----------------
[C:   QPI][CR:    0][CC:    0][L:   3]

QPI:		NAK (Length: 3)
[C:  QVFW][CR:    0][CC:    0][L:   3]

QVFW:		NAK (Length: 3)

[C: ^P005PI][CR:    0][CC:    0][L:   3]

^P005PI:		NAK (Length: 3)

[C:   QPI][CR:    0][CC:    0][L:   3]

[C:  QVFW][CR:    0][CC:    0][L:   3]

QVFW:		NAK (Length: 3)

[C: ^P005PI][CR:    0][CC:    0][L:   3]

^P005PI:		NAK (Length: 3)

[C:   QPI][CR:    0][CC:    0][L:   3]

QPI:		NAK (Length: 3)
[C:  QVFW][CR:    0][CC:    0][L:   3]

QVFW:		NAK (Length: 3)

[C: ^P005PI][CR:    0][CC:    0][L:   3]

^P005PI:		NAK (Length: 3)

Invalid Slave ID

Invalid Slave ID

Time-out
Invalid Slave ID
Invalid Slave ID

Time-out

Invalid Slave ID

Invalid Slave ID

Time-out

Invalid Slave ID

Invalid Slave ID
[C:   QPI][CR:    0][CC:    0][L:   3]

QPI:		NAK (Length: 3)

[C:  QVFW][CR:    0][CC:    0][L:   3]

QVFW:		NAK (Length: 3)

[C: ^P005PI][CR:    0][CC:    0][L:   3]

^P005PI:		NAK (Length: 3)

Try Autodetect Protocol

[C:   QPI][CR:    0][CC:    0][L:   3]

QPI:		NAK (Length: 3)
```

softwarecrash, 2024-10-21:
thats curios. so i send the same QVFW comand as the watchpower but your device answer with NAK
lets try with another comand
[Solar2MQTT_1.2.0-Pre5A4_OTA.bin.zip](https://github.com/user-attachments/files/17466228/Solar2MQTT_1.2.0-Pre5A4_OTA.bin.zip)

avenseElectro, 2024-10-21:
> thats curios. so i send the same QVFW comand as the watchpower but your device answer with NAK lets try with another comand [Solar2MQTT_1.2.0-Pre5A4_OTA.bin.zip](https://github.com/user-attachments/files/17466228/Solar2MQTT_1.2.0-Pre5A4_OTA.bin.zip)

That's strange...
```

[WebSerial] Connecting...
[WebSerial] Connected...
[C:  QMOD][CR:    0][CC:    0][L:   3]

QMOD:		 NAK (Length: 3)

[C: ^P005PI][CR:    0][CC:    0][L:   3]

^P005PI:		NAK (Length: 3)

Try Autodetect Protocol

[C:   QPI][CR:    0][CC:    0][L:   3]

QPI:		NAK (Length: 3)

[C:  QMOD][CR:    0][CC:    0][L:   3]

QMOD:		 NAK (Length: 3)

[C: ^P005PI][CR:    0][CC:    0][L:   3]

^P005PI:		NAK (Length: 3)

Try Autodetect Protocol

[C:   QPI][CR:    0][CC:    0][L:   3]

QPI:		NAK (Length: 3)

```seems to interrupt

softwarecrash, 2024-10-21:
ok your device speaks a different language, it answer on all with NAK
so last try is to get a raw communication snipet from your com port of the pc, it exsist some software that can do that

avenseElectro, 2024-10-21:
> ok your device speaks a different language, it answer on all with NAK yes that interrupt, have some exception that i dont find, its a bug for later. so last try is to get a raw communication snipet from your com port of the pc, it exsist some software that can do that
> 
> edit: can you see the com baudrate in watchpower? edit2: you can try this https://github.com/syssi/esphome-smg-ii/ its modbus variant, just for try, or pipsolar.

I've tried that one and it didn't work either.
19200
[image]

softwarecrash, 2024-10-21:
> > ok your device speaks a different language, it answer on all with NAK yes that interrupt, have some exception that i dont find, its a bug for later. so last try is to get a raw communication snipet from your com port of the pc, it exsist some software that can do that
> > edit: can you see the com baudrate in watchpower? edit2: you can try this https://github.com/syssi/esphome-smg-ii/ its modbus variant, just for try, or pipsolar.
> 
> I've tried that one and it didn't work either. 19200 ![Captura de ecrã 2024-10-21 212600](https://private-user-images.githubusercontent.com/166903640/378587967-bb0a5410-57b1-4ae2-8a4f-fcee56ec8eac.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk1NDI4MjYsIm5iZiI6MTcyOTU0MjUyNiwicGF0aCI6Ii8xNjY5MDM2NDAvMzc4NTg3OTY3LWJiMGE1NDEwLTU3YjEtNGFlMi04YTRmLWZjZWU1NmVjOGVhYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAyMVQyMDI4NDZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00MDNlMGIwOWM1ZjM1YWNjOTA0ZjQ0OTNlMmVhMjMyZTQ2ZDM0MGE1YzIwNDlmYjU3MDdjZGYyNzhkM2ExMWFkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.RNP5NhEAQ_KaH11uATHlp8qHRK6SHjY8sfbSZEOwF7o)

wait ... the default speed for PI protocols is 2400, so thats fucking fast. let me build short a version that changes to 19200

softwarecrash, 2024-10-21:
changes on QVFW querry to 19200
[Solar2MQTT_1.2.0-Pre5A5_OTA.bin.zip](https://github.com/user-attachments/files/17466680/Solar2MQTT_1.2.0-Pre5A5_OTA.bin.zip)

softwarecrash, 2024-10-21:
@all-solutions need your assitance here for watchpower and com monitoring

avenseElectro, 2024-10-21:
> changes on QVFW querry to 19200 [Solar2MQTT_1.2.0-Pre5A5_OTA.bin.zip](https://github.com/user-attachments/files/17466680/Solar2MQTT_1.2.0-Pre5A5_OTA.bin.zip)
> 
> dont know its a typo error or right, at the top of it stand modbus settings, so its possible that you have a version that speaks modbus

Unfortunately it didn't work either....

I would suggest removing this inverter model from your "inverters that work"

softwarecrash, 2024-10-21:
> Unfortunately it didn't work either....
>
> I would suggest removing this inverter model from your "inverters that work"

not so fast ;)
so[ here someone wrote](https://rain.linuxoid.in/2024/08/30/%D0%B0%D0%B2%D1%82%D0%BE%D0%BD%D0%BE%D0%BC%D0%BA%D0%B0-v1/) that a device with this name works, so your device is different from other with same name, wait for all-sollutions, he can explain you how to get raw data from the com port comunication, with this data we can clearly see whats going on with your device

avenseElectro, 2024-10-21:
> > > changes on QVFW querry to 19200 [Solar2MQTT_1.2.0-Pre5A5_OTA.bin.zip](https://github.com/user-attachments/files/17466680/Solar2MQTT_1.2.0-Pre5A5_OTA.bin.zip)
> > > dont know its a typo error or right, at the top of it stand modbus settings, so its possible that you have a version that speaks modbus
> > 
> > 
> > Unfortunately it didn't work either....
> > I would suggest removing this inverter model from your "inverters that work"
> 
> try to aks your seller for the comunication protocol, thtas was helpfull too
> 
> not so fast ;) so[ here someone wrote](https://rain.linuxoid.in/2024/08/30/%D0%B0%D0%B2%D1%82%D0%BE%D0%BD%D0%BE%D0%BC%D0%BA%D0%B0-v1/) that a device with this name works, so your device is different from other with same name, wait for all-sollutions, he can explain you how to get raw data from the com port comunication, with this data we can clearly see whats going on with your device

Ok, I've already asked but I doubt I'll get an answer...
In this article he mentions that it didn't work...

softwarecrash, 2024-10-21:
> Ok, I've already asked but I doubt I'll get an answer... In this article he mentions that it didn't work...

at the bottom its a hint that it work, its the only thing i found on search your device name with qpigs or qpiri (protocol comands)

avenseElectro, 2024-10-23:
@softwarecrash the cable that came with inverter has 4 connections....
[image]
[image]
[image]

softwarecrash, 2024-10-23:
> @softwarecrash the cable that came with inverter has 4 connections....
> [image]
> [image]
> [image]
> 

2 for Data
2 for Power, Take a Look at the Wiki and Protocol folder

avenseElectro, 2024-10-23:
@softwarecrash 
tx
rx
12v
gnd
???????

avenseElectro, 2024-10-23:
> Your cable is for the wifi dongle. That's why you have also the 12v pin in that.
> 
> BTW: Can you please show us the complete wiring from your inverter to the Wemos? Perhaps there is a problem. We can not solve it, when we don't know your wiring.
> 
> > show-me wlki link please
> 
> Really??? ![image](https://private-user-images.githubusercontent.com/17761850/379422529-b245ecc6-9f45-4a49-abf9-eed1812a5ded.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk3MTA1MTEsIm5iZiI6MTcyOTcxMDIxMSwicGF0aCI6Ii8xNzc2MTg1MC8zNzk0MjI1MjktYjI0NWVjYzYtOWY0NS00YTQ5LWFiZjktZWVkMTgxMmE1ZGVkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDIzVDE5MDMzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJmYWI2OTI4NTY1NDZkYmVjYjBhNzI1OWFjNTBiNGY2NDA3OTA1MjM2YmU4YjRhNDYyMjhmZmYxNDhjOGJiNWImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.kIq_q8EOWAW2UtVZzepb_96B_kZOnBh_akoBaP8Euuo)

If my connection was wrong, the inverter I have that uses the pip30 wouldn't work either...

But here's my wiring:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

avenseElectro, 2024-10-23:
In the photo it shows pins D1 and D2 but it was to test the esphome at the moment it is on D6 and D7

avenseElectro, 2024-10-24:
>You definitely have an SMH, not an SMG?

[image]

madmaximus, 2024-11-01:
Almost the same problem with my 2.2kW EASun
[image]

Looks that my esp-01 could not recognize protocol

softwarecrash, 2024-11-01:
> Almost the same problem with my 2.2kW EASun ![Screenshot_2024-11-01-06-12-41-432_com android chrome](https://private-user-images.githubusercontent.com/741928/382182689-b0c498dc-1eb5-47c7-9045-63081b879841.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA0NjE2MjQsIm5iZiI6MTczMDQ2MTMyNCwicGF0aCI6Ii83NDE5MjgvMzgyMTgyNjg5LWIwYzQ5OGRjLTFlYjUtNDdjNy05MDQ1LTYzMDgxYjg3OTg0MS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTAxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEwMVQxMTQyMDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02Njk4MDIzZDgzYjUwNzYyY2U5NjA0NTAxZjQ1MTIyMjVhMjM0MjY1MjUyNzcwN2YzZDdmNjZlNmM4N2E4OThjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.Vz48yCjeHkw5WhTIEZUm3ekzxkFCRaGx_Alut-rCT7E)
> 
> Looks that my esp-01 could not recognize protocol

What for easun Model?

madmaximus, 2024-11-01:
[image]
This marking on the side

madmaximus, 2024-11-02:
> ### A clear and concise description of what the problem is.
> I can't get this inverter to work at all, can anyone help me?
>
> ```
>
> [WebSerial] Connecting...
> [WebSerial] Connected...
> [C: ^P005PI][CR: [...]

I found the solution it seems rx and tx wire wrong marking, so i just replace one to another and its works !!!
Try itz because my errors was similar 
Now i have connection 😄

miggi, 2024-12-02:
The same situation for me, with EASUN SMG-II 6.2Kw, cannot make it to work. Also tried your latest recommendation to switch tx/rx, but without result. 

Maybe it's connectivity issue ? No idea how to read that trobuleshooting log :)

-----
Response Timed Out
Time-out
Request failed!
----------------- End Autodetect -----------------
ProzessData P:0 C:0
MQTT Client State: 0
Data sent to MQTT
----------------- Start Autodetect -----------------
Try Autodetect Protocol
[C:    QPI][CR:    0][CC:    0][L:   3]
QPI:		NOA (Length: 3)
[C: ^P005PI][CR:    0][CC:    0][L:   3]
^P005PI:		NOA (Length: 3)
Try Autodetect Protocol
[C:    QPI][CR:    0][CC:    0][L:   3]
QPI:		NOA (Length: 3)
[C: ^P005PI][CR:    0][CC:    0][L:   3]
^P005PI:		NOA (Length: 3)
Try Autodetect Protocol
[C:    QPI][CR:    0][CC:    0][L:   3]
QPI:		NOA (Length: 3)
[C: ^P005PI][CR:    0][CC:    0][L:   3]
^P005PI:		NOA (Length: 3)
Try Autodetect Modbus device
Init MUST PV/PH 18 protocol, baud 19200, modbusAddr 4

closed 2025-01-17 by github-actions[bot]