softwarecrash/Solar2MQTT-ESP8266 #164: [Bug]: no PV2_Input_Voltage sensor on homeassistant

vpm, 2024-10-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

no PV2_Input_Voltage sensor on homeassistant, but data exist

### Screenshots / Fotos

[image]
[image]
[image]

### Version

1.x.x and above

### Relevant livejson output

```json
{
  "EspData": {

  },
  "DeviceData": {
    "AC_in_rating_voltage": 230,
    "AC_in_rating_current": 31.3,
    "AC_out_rating_voltage": 230,
    "AC_out_rating_frequency": 50,
    "AC_out_rating_current": 31.3,
    "AC_out_rating_apparent_power": 7200,
    "AC_out_rating_active_power": 7200,
    "Battery_rating_voltage": 48,
    "Battery_re-charge_voltage": 49,
    "Battery_under_voltage": 48,
    "Battery_bulk_voltage": 57.6,
    "Battery_float_voltage": 57.6,
    "Battery_type": 3,
    "Current_max_AC_charging_current": 2,
    "Current_max_charging_current": 70,
    "Input_voltage_range": "UPS",
    "Output_source_priority": "SBU first",
    "Charger_source_priority": "Only solar charging permitted",
    "Parallel_max_num": 9,
    "Machine_type": "Off Grid",
    "Topology": "Transformerless",
    "Output_mode": 0,
    "Battery_re-discharge_voltage": 50,
    "PV_OK_condition_for_parallel": 0,
    "PV_power_balance": 1,
    "Max_charging_time_at_CV_stage": 900,
    "Operation_Logic": 0,
    "Max_discharging_current": 0
  },
  "LiveData": {

  }
}
```

### What browsers are you seeing the problem on?

Chrome

vpm, 2024-10-21:
{
  "EspData": {
    "Device_name": "solar2mqtt",
    "ESP_VCC": 3.041,
    "Wifi_RSSI": -45,
    "sw_version": "1.2.0-Pre3.1",
    "Free_Heap": 20352,
    "HEAP_Fragmentation": 3,
    "json_memory_usage": 1451,
    "json_capacity": 2048,
    "runtime": 87,
    "ws_clients": 0,
    "detect_protocol": 2
  },
  "DeviceData": {
    "AC_in_rating_voltage": 230,
    "AC_in_rating_current": 31.3,
    "AC_out_rating_voltage": 230,
    "AC_out_rating_frequency": 50,
    "AC_out_rating_current": 31.3,
    "AC_out_rating_apparent_power": 7200,
    "AC_out_rating_active_power": 7200,
    "Battery_rating_voltage": 48,
    "Battery_re-charge_voltage": 49,
    "Battery_under_voltage": 48,
    "Battery_bulk_voltage": 57.6,
    "Battery_float_voltage": 57.6,
    "Battery_type": 3,
    "Current_max_AC_charging_current": 2,
    "Current_max_charging_current": 70,
    "Input_voltage_range": "UPS",
    "Output_source_priority": "SBU first",
    "Charger_source_priority": "Only solar charging permitted",
    "Parallel_max_num": 9,
    "Machine_type": "Off Grid",
    "Topology": "Transformerless",
    "Output_mode": 0,
    "Battery_re-discharge_voltage": 50,
    "PV_OK_condition_for_parallel": 0,
    "PV_power_balance": 1,
    "Max_charging_time_at_CV_stage": 900,
    "Operation_Logic": 0,
    "Max_discharging_current": 0,
    "Device_Model": "MKS2-7200",
    "Protocol_ID": "PI30"
  },
  "LiveData": {
    "AC_in_Voltage": 218.5,
    "AC_in_Frequenz": 49.9,
    "AC_out_Voltage": 229.8,
    "AC_out_Frequenz": 49.9,
    "AC_out_VA": 1884,
    "AC_out_Watt": 1703,
    "AC_out_percent": 26,
    "Inverter_Bus_Voltage": 387,
    "Battery_Voltage": 52.7,
    "Battery_Charge_Current": 0,
    "Battery_Percent": 80,
    "Inverter_Bus_Temperature": 35,
    "PV_Input_Current": 0.2,
    "PV_Input_Voltage": 90,
    "Battery_SCC_Volt": 0,
    "Battery_Discharge_Current": 27,
    "Status_Flag": 10110,
    "Battery_voltage_offset_fans_on": 0,
    "EEPROM_Version": 0,
    "PV_Charging_Power": 18,
    "Device_Status": 10,
    "Battery_Load": -27,
    "PV_Input_Power": 0,
    "PV2_Input_Current": 0.9,
    "PV2_Input_Voltage": 303.3,
    "PV2_Charging_Power": 279,
    "PV2_Input_Power": 0,
    "Inverter_Operation_Mode": "Battery",
    "PV_generation_sum": 9208900,
    "PV_generation_year": 3776600,
    "PV_generation_month": 210500,
    "PV_generation_day": 17549,
    "AC_in_generation_sum": 10486200,
    "AC_in_generation_year": 4722700,
    "AC_in_generation_month": 492300,
    "AC_in_generation_day": 18635
  }
}

softwarecrash, 2024-10-21:
please try this
[Solar2MQTT_1.2.0-Pre5A2.zip](https://github.com/user-attachments/files/17465381/Solar2MQTT_1.2.0-Pre5A2.zip)

vpm, 2024-10-21:
fixed
th

[image]
[image]

closed 2024-10-21 by vpm