softwarecrash/Solar2MQTT-ESP8266 #171: [Bug]: Web Console stops working in few seconds

sidmeyer, 2024-11-01:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

The Web Console only shows few lines and then stops updating.
The following error appears in browser console: "webserial:1 WebSocket connection to 'ws://192.168.1.205/webserialws' failed: Could not decode a text frame as UTF-8."
Sometimes works more stably for ~1 minute.

Checked in Chrome (desktop and Android), Edge, Safari.

Solar2MQTT version: 1.1.8
Inverter model: PowMr POW-HVM3.2H-24V-N
Inverter firmware version: 1.0 27 009

### Screenshots / Fotos

[image]

### Steps To Reproduce

1. Open web ui
2. click Web Console
3. the same problem if refresh the Web console

### Version

1.x.x and above

### Relevant livejson output

```json
{"EspData":{"Device_name":"PowMr","ESP_VCC":2.958,"Wifi_RSSI":-68,"sw_version":"1.1.8","Free_Heap":23816,"HEAP_Fragmentation":17,"json_memory_usage":1209,"json_capacity":2048,"runtime":893,"ws_clients":0,"detect_protocol":2},"DeviceData":{"AC_in_rating_voltage":220,"AC_in_rating_current":13,"AC_out_rating_voltage":220,"AC_out_rating_frequency":50,"AC_out_rating_current":13.6,"AC_out_rating_apparent_power":3000,"AC_out_rating_active_power":3000,"Battery_rating_voltage":24,"Battery_re-charge_voltage":25.5,"Battery_under_voltage":24,"Battery_bulk_voltage":27.6,"Battery_float_voltage":27,"Battery_type":"User","Current_max_AC_charging_current":20,"Current_max_charging_current":30,"Input_voltage_range":"Appliance","Output_source_priority":"SBU first","Charger_source_priority":"Only solar charging permitted","Parallel_max_num":1,"Machine_type":"Off Grid","Topology":"Transformerless","Output_mode":0,"Battery_re-discharge_voltage":26.5,"PV_OK_condition_for_parallel":0,"PV_power_balance":1,"Device_Model":"VMII-NXPW5KW","Protocol_ID":"PI30"},"LiveData":{"AC_in_Voltage":237.7,"AC_in_Frequenz":49.9,"AC_out_Voltage":219.8,"AC_out_Frequenz":50,"AC_out_VA":222,"AC_out_Watt":202,"AC_out_percent":6,"Inverter_Bus_Voltage":386,"Battery_Voltage":25.92,"Battery_Charge_Current":0,"Battery_Percent":82,"Inverter_Bus_Temperature":37,"PV_Input_Current":0.1,"PV_Input_Voltage":25.6,"Battery_SCC_Volt":0,"Battery_Discharge_Current":8,"Status_Flag":10010,"Battery_voltage_offset_fans_on":0,"EEPROM_Version":0,"PV_Charging_Power":0,"Device_Status":10,"Battery_Load":-8,"PV_Input_Power":0,"Inverter_Operation_Mode":"Battery"}}
```

### What browsers are you seeing the problem on?

Chrome

softwarecrash, 2024-11-02:
Looks Like the ESP is Crashing, Just try the pre release

vpm, 2024-11-06:
i have same trouble with my first builded esp, but i make new from same components and new worked fine.

china quality :)

closed 2024-12-22 by github-actions[bot]