softwarecrash/Solar2MQTT-ESP8266 #172: [DEVICE_ADD] ANENJI ModBus

softwarecrash, 2024-11-04:
### A clear and concise description of what the problem is.

asdasdasdasdasd

### Additional context

asdasdasd

softwarecrash, 2024-11-05:
> added some basic support for ANENJI according to the pdf registers map [bigdigital@3230425](https://github.com/bigdigital/Solar2MQTT/commit/32304256e936716b9a5782027ee25a023019b351)

wow great... need some PR asap to avoid more code conflicts

lenivetsDCT, 2024-11-13:
Hi, i'm very interested in this project and willing to help.
I was trying to use it for Anenji 6.2 (blue one, no parallel)
Seams it's clone of EASUN SMG II

I couldn't use code from bigdigital (compiled and upload to Wemos D1 mini)
Using ESPHome component - https://github.com/syssi/esphome-smg-ii worked fine, but it's laggy and not stable (i spent 3h on configurating yaml... not to face OOM)

https://github.com/softwarecrash/Solar2MQTT/pull/174/files - created PR with @bigdigital (main kudos to Artem !) and added more info from provided pdf.

softwarecrash, 2024-11-13:
> Hi, i'm very interested in this project and willing to help.
> I was trying to use it for Anenji 6.2 (blue one, no parallel)
> Seams it's clone of EASUN SMG II
> 
> I couldn't use code from bigdigital (compiled and upload to Wemos D1 mini)
> Using ESPHome component - https://github.com/syssi/esphome-smg-ii worked fine, but it's laggy and not stable (i spent 3h on configurating yaml... not to face OOM)
> 
> 
> https://github.com/softwarecrash/Solar2MQTT/pull/174/files - created PR with @bigdigital (main kudos to Artem !) and added more info from provided pdf.
> 
> 
> here is my esphome config:
> ```
> substitutions:
>   name: anenji
>   friendly_name: anenji
>   tx_pin: GPIO13
>   rx_pin: GPIO12
> 
> esphome:
>   name: ${name}
>   friendly_name: ${friendly_name}
>   min_version: 2024.6.0
>   name_add_mac_suffix: false
>   project:
>     name: "syssi.esphome-smg-ii"
>     version: 1.2.0
> 
> esp8266:
>   board: d1_mini
> 
> wifi:
>   ssid: !secret wifi_ssid
>   password: !secret wifi_password
> 
> logger:
>   level: INFO
> 
> # If you use Home Assistant please remove this `mqtt` section and uncomment the `api` component!
> # The native API has many advantages over MQTT: https://esphome.io/components/api.html#advantages-over-mqtt
> # mqtt:
> #   broker: !secret mqtt_host
> #   username: !secret mqtt_username
> #   password: !secret mqtt_password
> #   id: mqtt_client
> #   discovery: False
> #   discover_ip: True
> 
> api:
>   encryption:
>     key: !secret api_key
> 
> ota:
>   - platform: esphome
>     password: !secret ota_pass
> 
> uart:
>   - id: uart_0
>     baud_rate: 9600
>     tx_pin: ${tx_pin}
>     rx_pin: ${rx_pin}
> 
> modbus:
>   - id: modbus0
>     uart_id: uart_0
>     send_wait_time: 200ms
> 
> modbus_controller:
>   - id: smg0
>     address: 0x01
[... 1048 lines cut ...]
>   #   max_value: 900.0
>   #   step: 1
>   #   unit_of_measurement: "min"
> 
>   # # Two equalization charging intervals                           day     Uint  337 1 R/W Range: 1~90
>   # - platform: modbus_controller
>   #   modbus_controller_id: smg0
>   #   name: "${name} two equalization charging intervals"
>   #   use_write_multiple: true
>   #   address: 337
>   #   register_type: holding
>   #   value_type: U_WORD
>   #   min_value: 0.0
>   #   max_value: 90.0
>   #   step: 1
>   #   unit_of_measurement: "day"
> 
> # Button entities
> ```

Hello,
Thanks for your Work, but badly its for the wrong Branch.
Actual i Work on the rework Branch and Change a Lot of Things.
So the whole Thing from bigdigital needs to implement in this Branch.
When you Like to do this you can Join the discord Server and write ne for more Details.

closed 2024-12-29 by github-actions[bot]

BieleckiLtd, 2025-05-15:
> [RS232 Communication Protocol V1.0-220209.pdf](https://github.com/user-attachments/files/17615615/RS232.Communication.Protocol.V1.0-220209.pdf)

Hi @softwarecrash, how did you get hold of this file? I’m looking for something like that for an Anenji ANJ-HHS-11 kW 48 V Wi-Fi.

softwarecrash, 2025-05-15:
> Hi [@softwarecrash](https://github.com/softwarecrash), how did you get hold of this file? I’m looking for something like that for an Anenji ANJ-HHS-11 kW 48 V Wi-Fi.

ask the vendor or seller

BieleckiLtd, 2025-05-19:
> ask the vendor or seller

I did, and they sent me one - similar to yours but with a different timestamp. Almost none of the registers matched 😔

Anyway, I ended up reverse engineering it myself. I’ll publish the map once I’ve confirmed a few things after hooking up the battery.