softwarecrash/Solar2MQTT-ESP8266 #173: [Bug]: Incorrect replies

SapozhnikUA, 2024-11-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

I need to read data from QPIRI.
If I set command "QPIGS" - everything Ok
But if I send command "MUCHGC002" I see in QPIRI answer "NAK" and instead Model Name (in QMN) I see data from QPIRI.
I tried  to record short video which shows this problem 

Perhaps I did something wrong :) 

### Screenshots / Fotos

https://github.com/user-attachments/assets/68ab99e2-0474-4e21-aa86-a24085718851

### Version

1.x.x and above

### Relevant livejson output

```json
{
"EspData": {
"Device_name": "PowMR_Invertor",
"ESP_VCC": 2.952,
"Wifi_RSSI": -71,
"sw_version": "1.1.8",
"Free_Heap": 23584,
"HEAP_Fragmentation": 8,
"json_memory_usage": 951,
"json_capacity": 2048,
"runtime": 84275,
"ws_clients": 0,
"detect_protocol": 2
},
"DeviceData": {
"Device_Model": "VMII-NXPW5KW",
"Protocol_ID": "PI30"
},
"LiveData": {
"AC_in_Voltage": 224.5,
"AC_in_Frequenz": 50,
"AC_out_Voltage": 229.7,
"AC_out_Frequenz": 49.9,
"AC_out_VA": 206,
"AC_out_Watt": 104,
"AC_out_percent": 5,
"Inverter_Bus_Voltage": 398,
"Battery_Voltage": 26.6,
"Battery_Charge_Current": 0,
"Battery_Percent": 80,
"Inverter_Bus_Temperature": 42,
"PV_Input_Current": 0,
"PV_Input_Voltage": 0,
"Battery_SCC_Volt": 0,
"Battery_Discharge_Current": 5,
"Status_Flag": 10000,
"Battery_voltage_offset_fans_on": 0,
"EEPROM_Version": 0,
"PV_Charging_Power": 0,
"Device_Status": 10,
"Battery_Load": -5,
"PV_Input_Power": 0,
"Inverter_Operation_Mode": "Battery",
"Time_until_absorb_charge": 0,
"Time_until_float_charge": 0,
"Tracker_temperature": 0,
"Inverter_temperature": 37,
"Battery_temperature": 30,
"Transformer_temperature": 42,
"Fan_speed": 1,
"Inverter_charge_state": "No charging"
}
}
```

### What browsers are you seeing the problem on?

no Issue with the Browser or WebUI

softwarecrash, 2024-11-10:
hello,
that looks like some request comands going wrong, but i cant rebuild it here at the moment.
i remember some bugs with muchcg so just try  "MUCHGC02" some devices struggle with leading zeros

SapozhnikUA, 2024-11-10:
Hey @softwarecrash,

Thanks for your reply, 
I installed pre-release version, it also has the same problems, but it has new data which needed to me anв I can ignore this BUG.  :) 

Perhaps, I found one more problem with answer but I don't know on which side. 
If I send request QMUCHGCR - as I understand, I will get back data with list of all values (Enquiry selectable value about max utility charging current). I receive them (002 010 020 030 040 050 060) but this is not full list. In my case I have also 70,80,100
Someone cut string, invertor or our device

Thanks!

softwarecrash, 2024-11-10:
> Hey @softwarecrash,
>
> Thanks for your reply,
> I installed pre-release version, it also has the same problems, but it has new data which needed to me anв I can ignore this BUG.  :)
>
> Perhaps, I found [...]

This is correct when it Shows until 60 in the RAW section or in the command answer, nothing ist Cut.
Your device is able to Charge max 60A from solar OR from grid, for the max whole Charge current you have another comand.
So the Bug with Set, try it with Digits 0002 (the First digit IS the Inverter number, so from ever value add a Zero.

SapozhnikUA, 2024-11-10:
> Your device is able to Charge max 60A from solar OR from grid, 

My device is able to charge more 60A from utility and it shows all these values on the his screen
For example I set 100A - charging current but QMUCHGCR return me only 2...60

[image]

softwarecrash, 2024-11-10:
MUCHGC is for max solar charging xurrent, not overall charging, so your device can max solar charge at 60A
grid charge command is request: QMCHGCR and set :MNCHGC
refering to https://github.com/softwarecrash/Solar2MQTT/blob/master/Protocol/PI30MAX.Communication.Protocol20210217.pdf

SapozhnikUA, 2024-11-10:
))) 
I think you are not right :) 

for my invertor 4.2kWt:
QMCHGCR - **M**aximum **CH**ar**G**ing  **C**u**R**ent = solar + utility  -> [10-120A]

QMUCHGCR (letter **U** means **U**tility)   **M**aximum **U**tility **CH**ar**G**ing  **C**u**R**ent  -> [2-100A]

If I set Current_max_charging_current = 110A and Current_max_AC_charging_current =100A,  I can get from solar only 10A and 100 from utility
ps. Really,  I don't have solar at all.

[image]

[image]

Both these values I correctly rad and set by commands:
[image]

But both queries  QMCHGCR and QMUCHGCR return me back not full arrays:

[image]

[image]

Have a nice night and tanks for your answers!

softwarecrash, 2024-11-10:
sure the U is utility, without U is whole charge. sometimes i mixed it up in my head XD

let me check this the next days what limit the answer strings.
just for fun, when you have the pc cable you can use watchpower and enable debug to check the real answers

softwarecrash, 2024-11-11:
so just checked with the emulator, the code can handle long answers
[image]

closed 2024-12-27 by github-actions[bot]