softwarecrash/Solar2MQTT-ESP8266 #178: New working device

nomike2004, 2024-11-24:
### Please enter the name of the device manufacturer

easun

### Please enter the device type

inverter

softwarecrash, 2024-12-01:
can you provide a picture of the model label?

closed 2024-12-07 by softwarecrash

nomike2004, 2024-12-23:
Guten Abend,über Ihren Shop habe ich einen Fertigbausatz Solar2Mqtt gekauft.(Bestellnummer #24-0917) (8. Dezember 2024)Können Sie mir bitte schreiben wie der RJ45 Anschluss belegt ist?In der erhalten PDF steht leider dazu nichts, ich möchte mir das passende Kabel crimpen.Ich freue mich über eine Rückmeldung.Michael FeltlMessage ID: ***@***.***>

nomike2004, 2024-12-23:
Danke für die schnelle Antwort.Ein Inverter Easun SMR II 11kW soll angeschlossen werden.Hatte vorher einen anderen Easun Inverter und mit der freien Software "Solaranzeige" und über mqtt mit dem Home Assistant betrieben."Bastle" schon seit Wochen herum und bringe den nicht zum Laufen, deshalb hat mich jetzt Ihre Lösung interessiert.Gesendet mit der mobilen Mail AppAm 23.12.24 um 21:28 schrieb ManuA
                
            
            
                
                    Von: "ManuA" ***@***.***>Datum: 23. Dezember 2024An: "softwarecrash/Solar2MQTT" ***@***.***>Cc: "Author" ***@***.***>,"nomike2004" ***@***.***>Betreff: Re: [softwarecrash/Solar2MQTT] New working device (Issue #178)
                
                
Was für ein Inverter ist denn angeschlossen bzw. soll angeschlossen werden?
Grundsätzlich ist die Buchse an der Platine so belegt, dass diese mit einem ganz normalen Netzwerkkabel verbunden werden kann.

nomike2004, 2024-12-23:
Nein, das ist schon in Ordnung auch wenn das Gerät nicht mit dem Inverter funktioniert.Das wird es wahrscheinlich sein, dass das Protokoll nicht mehr stimmt.Bin über Youtube auf Solar2MQTT gestoßen und fand es großartig, dass es sozusagen ein "freies" Projekt ist. Habe dann zum Herumlöten angefangen und mich dann aber rasch für die fertige Platine entschieden.Vielen herzlichen Dank für Ihr Engagement.Wünsche Ihnen noch schöne Feiertage.Michael Feltl Gesendet mit der mobilen Mail AppAm 23.12.24 um 22:43 schrieb ManuA
                
            
            
                
                    Von: "ManuA" ***@***.***>Datum: 23. Dezember 2024An: "softwarecrash/Solar2MQTT" ***@***.***>Cc: "nomike2004" ***@***.***>,"Author" ***@***.***>Betreff: Re: [softwarecrash/Solar2MQTT] New working device (Issue #178)
                
                
EaSun bastelt in letzter Zeit sehr an deren Invertern. Bisher war nur bekannt, dass die SMG-Serie das Modbus-Protokoll nutzt. Aber es tauchen immer mehr Serien von denen auf, die auf Modbus nach und nach scheinbar umgestellt werden.
Wir arbeiten zwar an der Implementierung von Modbus, aktuell ist das eaSun-Protokoll aber noch nicht implementiert. Und auch alles andere ist noch eine absolute Beta.
Wenn Du nicht so lange warten möchtest, meld Dich bitte per Mail im Shop. Wir klären dann den Rückversand und Erstattung.