softwarecrash/Solar2MQTT-ESP8266 #179: [DEVICE_ADD] POW-HVM6.2K-48V-LIP Add Modbus support

v-moto, 2024-11-25:
### Is your feature request related to a problem? Please describe

I cannot connect a new inverter to the HA

### Describe the solution you'd like

I received a description of the inverter's communication protocol from the manufacturer, please add minimal support for this device
[Modbus_RTU communication description.pdf](https://github.com/user-attachments/files/17908944/Modbus_RTU.communication.description.pdf)

### Describe alternatives you've considered

I did not find alternatives for this device, and it is not advisable to buy their adapter because it cannot be connected to HA

softwarecrash, 2024-11-26:
Let it Check later.
@bigdigital See you a Chance to build it in?

v-moto, 2024-12-01:
@softwarecrash @bigdigital 
Do you have a usdt trc20 wallet for donations? Because I really like your project and want to support you a little) and maybe it will speed up the adaptation of my inverter a little))

softwarecrash, 2024-12-01:
Badly No, PayPal ist linked in the Project for donation.
But sorry it Doenst Speed it Up, i have much Work on buissenes and less free time. So when i can get a Bit time i will Work on it.
Its Bit Tricky to Work without the device, i will try it when find some time

v-moto, 2024-12-01:
Okay.
Then I'll send it via Paypal one of these days. 
Thank you very much)
If you need any help with the tests, I am always happy to help

bigdigital, 2024-12-02:
@v-moto pls contact me in discord so i can send you a test version

sMusia, 2024-12-02:
Hi, I’m also an owner of such an inverter. I can join the tests too!

v-moto, 2024-12-02:
> Hi, I’m also an owner of such an inverter. I can join the tests too!

I have already received the beta, today I will test it, if everything goes well, I will write the result

v-moto, 2024-12-03:
> Hi, I’m also an owner of such an inverter. I can join the tests too!

Where are you from? And do you have an original adapter for this inverter?

sMusia, 2024-12-03:
> > Hi, I’m also an owner of such an inverter. I can join the tests too!
> 
> Where are you from? And do you have an original adapter for this inverter?

I am from Ukraine. Unfortunately, I don’t have the original adapter. Here's the story: I had a POW-HVM-3.5H-24V inverter. At that time, there was no information on the internet about the communication protocol with the inverter, so I independently developed a way to retrieve some telemetry for HomeAssistant. Later, I switched to ESPHOME, which provided more telemetry data. Now, I’ve upgraded to the POW-HVM6.2K-48V-LIP and discovered the com port (RS232) and RJ45 (RS485). I couldn’t establish communication with either port—the inverter simply didn’t respond. That’s when I started thinking that the folks at POWMR might have changed the communication protocol. Then I came across Solar2MQTT and this query...

essergey, 2024-12-20:
@sMusia How did you manage to get the data?

tsdaemon, 2024-12-30:
> Figured out the registers and made a working version based on ESPHOME. ![image](https://private-user-images.githubusercontent.com/40174981/396062948-3ef577ac-6470-42e9-b58e-cf380feada3a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU1Nzk0NjgsIm5iZiI6MTczNTU3OTE2OCwicGF0aCI6Ii80MDE3NDk4MS8zOTYwNjI5NDgtM2VmNTc3YWMtNjQ3MC00MmU5LWI1OGUtY2YzODBmZWFkYTNhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjMwVDE3MTkyOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRiMDJkM2YzZjEyMGExN2VlYmY4NjNkN2VmMzQ0MWM0ZTE5YjE1OGI0MmEwYmQ3ZGIxZTM0Y2EyOTIxMjVjYjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.CpbiRsw4u_5R6C05VsQdx7rms2Wtc9vG4lO2OBEUWg0)

Hi! Thank you for sharing, can you share more details how did you do this with ESPHOME? I am also from Ukraine, and I am trying to set the monitoring for the same inverter.

sMusia, 2024-12-31:
> > Figured out the registers and made a working version based on ESPHOME.
> 
> Hi! Thank you for sharing, can you share more details how did you do this with ESPHOME? I am also from Ukraine, and I am trying to set the monitoring for the same inverter.

[inverter-powmr.zip](https://github.com/user-attachments/files/18280265/inverter-powmr.zip)

I used ESP8266, connection: esp: tx_pin: GPIO1, rx_pin: GPIO3 -> (RS-232 to UART, MAX3232) -> com_port: pin 2 - TX, pin 3 - RX, pin 5 - GND, pin 9 - +12-15V (with a step-down power module for ESP8266 and MAX3232). 

ESPHOME yaml in attachment.

Very important!: Make as few requests as possible on the Modbus bus, refer to - Optimizing modbus communications - https://esphome.io/components/modbus_controller

gyp5y, 2025-01-18:
Hi there!
I've just got data from my POW-HVM6.2K-48V-LIP using sample provided by @sMusia . It's great! Thank you!
The only thing that looks strange it's "Working Mode" values. I've got 8 and 9 on mine. Do you have extended description for those modes by any chance?

Fazabein, 2025-01-23:
Hello, my friend, can you make a tutorial on how to make a BMS cable? and which one should lead to the battery and which one should lead to the inverter? I use a hybrid inverter POW-HVM6.2K-48V-LIP and 2 amsan 48v 100ah wall mounted batteries 

[image]

. Can you help me ??

gyp5y, 2025-01-23:
I use pin-to-pin straight RJ-45 cable connected to RS-485 port on BMS and inverter but your case could be different if your batteries aren't based on JK-BMS hardware. Then I was playing with protocol selection on BMS and on inverter. I don't remember  which one I'm using. I will check in a few days and write it here.