softwarecrash/Solar2MQTT-ESP8266 #180: [Bug]: Solar2MQTT 2.0.0-Pre6: MQTT Topics doppelt vorhanden (Groß-/Kleinschreibung)

rolsch, 2024-11-26:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Wemos D1 Mini

### What happened?

Nach der Installation der Solar2MQTT 2.0.0-Pre6 haben sich die MQTT Einträge gedoppelt.

Beispiel

Inverter
EA SUN Power, SV-IV-5.6KW

Proto: P18

MQTT in IOBroker

Beispiel
neuer Eintrag (Anmerkung: dieser Wert wird nun auch aktualisiert)
SolarEASun/LiveData/Fan_Speed

alter Eintrag (dieser Wert wurde bisher nie befüllt/aktualisier)
SolarEASun/LiveData/Fan_speed

### Screenshots / Fotos

[image]

### Version

1.x.x and above

### Relevant livejson output

```json
{
  "EspData": {
    "Device_name": "Solar2MQTT",
    "ESP_VCC": 2.956,
    "Wifi_RSSI": -46,
    "sw_version": "2.0.0-Pre6",
    "Free_Heap": 11504,
    "HEAP_Fragmentation": 21,
    "runtime": 661,
    "ws_clients": 1,
    "detect_protocol": 1,
    "detect_raw_qpi": "18"
  },
  "DeviceData": {
    "AC_In_Rating_Voltage": 220,
    "AC_In_Rating_Current": 25.4,
    "AC_Out_Rating_Voltage": 220,
    "AC_Out_Rating_Frequency": 50,
    "AC_Out_Rating_Current": 25.4,
    "AC_Out_Rating_Apparent_Power": 5600,
    "AC_Out_Rating_Active_Power": 5600,
    "Battery_Rating_Voltage": 48,
    "Battery_Recharge_Voltage": 46,
    "Battery_Redischarge_Voltage": 50,
    "Battery_Under_Voltage": 46,
    "Battery_Bulk_Voltage": 57,
    "Battery_Float_Voltage": 57,
    "Battery_Type": 8,
    "Current_Max_AC_Charging_Current": 2,
    "Current_Max_Charging_Current": 120,
    "Input_Voltage_Range": "UPS",
    "Output_Source_Priority": "Solar-Battery-Utility",
    "Charger_Source_Priority": "Only solar",
    "Parallel_Max_Num": 9,
    "Machine_Type": "Grid-Tie",
    "Topology": "Transformerless",
    "Output_mode": "Single module",
    "Solar_Power_Priority": "Load-Battery-Utility",
    "MPPT_String": 1,
    "Protocol_ID": 18,
    "Buzzer_Enabled": true,
    "Overload_Bypass_Enabled": true,
    "LCD_Reset_To_Default_Enabled": true,
    "Overload_Restart_Enabled": true,
    "Over_Temperature_Restart_Enabled": true,
    "LCD_Backlight_Enabled": false,
    "Primary_Source_Interrupt_Alarm_Enabled": true,
    "Record_Fault_Code_Enabled": false
  },
  "LiveData": {
    "AC_In_Voltage": 241.8,
    "AC_In_Frequenz": 50,
    "AC_Out_Voltage": 241.8,
    "AC_Out_Frequenz": 50,
    "AC_Out_VA": 3,
    "AC_Out_Watt": 3,
    "AC_Out_Percent": 0,
    "Battery_Voltage": 54,
    "Battery_SCC_Volt": 0,
    "Battery_SCC2_Volt": 0,
    "Battery_Discharge_Current": 0,
    "Battery_Charge_Current": 14,
    "Battery_Percent": 98,
    "Inverter_Bus_Temperature": 36,
    "MPPT1_Charger_Temperature": 0,
    "MPPT2_Charger_Temperature": 0,
    "PV1_Input_Power": 702,
    "PV2_Input_Power": 0,
    "PV1_Input_Voltage": 258.5,
    "PV2_Input_Voltage": 0,
    "Configuration_State": 0,
    "MPPT1_Charger_Status": 2,
    "MPPT2_CHarger_Status": 0,
    "Load_Connection": 1,
    "Battery_Power_Direction": 1,
    "ACDC_Power_Direction": 0,
    "Line_Power_Direction": 1,
    "Local_Parallel_ID": 0,
    "PV_Input_Voltage": 258,
    "PV_Charging_Power": 702,
    "PV_Input_Current": 2.71,
    "Battery_Load": 14,
    "Inverter_Operation_Mode": "Hybrid",
    "Time_Until_Absorb_Charge": 0,
    "Time_Until_Float_Charge": 8,
    "SCC_Flag": 0,
    "AllowSccOnFlag": 0,
    "Charge_Average_Current": 0,
    "Tracker_Temperature": 0,
    "Inverter_Temperature": 36,
    "Battery_Temperature": 24,
    "Transformer_Temperature": 34,
    "": 0,
    "Fan_Lock_Status": 0,
    "Fan_Speed": 30,
    "SCC_Charge_Power": 0,
    "Parallel_Warning": 0,
    "Sync_Frequency": 0,
    "Inverter_Charge_State": "Absorb",
    "Fault_Code": "Ok",
    "PV_Generation_Day": 0,
    "PV_Generation_Month": 0,
    "PV_Generation_Year": 740132,
    "PV_Generation_Sum": 0
  }
}
```

### What browsers are you seeing the problem on?

no Issue with the Browser or WebUI

softwarecrash, 2024-11-26:
Das ist kein Bug,
Steht auch in den Infos zu dem Release das eine Umbenennung stattgefunden hat um es zu vereinheitlichen.

rolsch, 2024-11-27:
Ok, die Info zur Umbenennung hatte ich nicht gesehen.

Unschön ist nun halt die Tatsache dass ich den Ordner löschen muss.

Durch die Umbenennung/Löschung gehen mir so alle History-Daten im ioBroker verloren
und ich muss die jeweiligen Datenpunkte in allen meinen Scripten usw. korrigieren...

- die Umbenennung bezüglich Groß/Klein hätte man auch so belassen können... :-(

closed 2024-11-27 by softwarecrash