softwarecrash/Solar2MQTT-ESP8266 #183: [HELP] Getting, Exception 28 (LoadProhibited) invalid access to eeprom error, when using similar OTA credentials update method for other project.

haadi76, 2024-11-28:
### A clear and concise description of what the problem is.

I know that this is not related to _"Solar2MQTT"_, but do you know why my nodemcu(esp8266) is giving eeprom exception error when I tried to integrate the same OTA credential update method like you did. I just want to ask, if you might have encountered similar error. I would be really grateful if you could suggest some possible solutions. 

I am using sinric pro to make smart switch of my nodemcu and I am trying to update the credential other that wifi over the ai, using ESPAyncWebserver library for the OTA server.

softwarecrash, 2024-11-28:
i cant help you with this, 28 say the pointed value is not there where the pointer. you must debug this and search wich value is moved or non exsist on the "eeprom"

haadi76, 2024-11-28:
ok , thanks