softwarecrash/Solar2MQTT-ESP8266 #185: [HELP] <title> Does not read data from the inverter axioma energy ismppt bf 3000

Dior34, 2024-12-01:
### A clear and concise description of what the problem is.

Does not read data from the inverter axioma energy ismppt bf 3000.
Several times it was possible to get data in the Home Assistant, this confirms that the settings are correct. In this case, the Solar2MQTT_d1_mini_1.2.0-Pre4.bin firmware was used. The data was obtained 3 times for 2-3 hours. The rest of the time, either does not read anything at all or 0
Firmware changes do not help.
at the same time, wachpower receives data only if the version is older than version 1.1.9. i.e. 1.1.10 does not receive 

### Additional context

[image]

[image]

Dior34, 2024-12-01:
[image]

softwarecrash, 2024-12-01:
What tells the webserial (or USB serial) from the ESP when No Data is recived?

Dior34, 2024-12-02:
I used different boards based on 8266mod. also differences in TTL boards from different manufacturers
[image]
[image]
[image]
[image]
[image]
[image]
[image]

There is a photo of a successful connection left
[image]

If the connection is successful, the PI30 inverter protocol is determined
There are still photos of various unsuccessful connections. I'll upload more later.
[image]
[image]
[image]
[image]

softwarecrash, 2024-12-03:
I have No Idea why it Work and dont Work later.
Looks Like a communication issue

Dior34, 2024-12-04:
Please tell me if it is possible to manually poll an unconnected inverter to check if it responds to requests.

softwarecrash, 2024-12-04:
you can try to force a protocol by paste following in the webserial console

setp x
  1=  PI18,
   2= PI30,
   3= MODBUS_MUST,
   4= MODBUS_DEYE,
   5= MODBUS_ANENJI,
   6= PROTOCOL_TYPE_MAX

Dior34, 2024-12-07:
Still managed to start.
Ordered a new (third) TTL board. And everything worked.
Can you add this model of hybrid inverter as working

Thanks!

closed 2024-12-07 by softwarecrash