softwarecrash/Solar2MQTT-ESP8266 #195: [FEATURE] Suggestion to update some data in MQTT more often

ivan-rio, 2024-12-26:
### Is your feature request related to a problem? Please describe

I created a few select controls for charging voltage, priority etc in Home Assistant using "DeviceData/..." MQTT topics as state topics for these controls. But the problem is that these topics are only updated on Solar2Mqtt startup. So i should manually reboot ESP after changing any parameter, to change the state of select controls.

### Describe the solution you'd like

It will be great if in Solar2Mqtt settings will be option to enable more frequent updates of "DeviceData/..." topics like in "LiveData" topics.

### Describe alternatives you've considered

The only solution now is to reboot Solar2Mqtt manually.

AlexK98, 2024-12-27:
@ivan-rio can you share your select controls somewhere?

ivan-rio, 2024-12-29:
> @ivan-rio can you share your select controls somewhere?

Here is short example:
`
mqtt:
  select:
      - name: "Float Voltage"
        icon: mdi:flash
        unique_id: "Easun SMH-II-0F6C29.select_float_voltage"
        options:
          - "27 V"
          - "28 V"
        state_topic: "Easun/DeviceData/Battery_float_voltage"
        value_template: >
          {% set values_map = {
                "27": "27 V",
                "28": "28 V",
              }
          %}
          {{ values_map[value] }}
        command_topic: "Easun/DeviceControl/Set_Command"
        command_template: >
          {% set value_map = {
                "27 V": "27.0",
                "28 V": "28.0",
              }
          %}
          PBFT{{ value_map[value] }}
        optimistic: false
        qos: 0
        retain: false
        device:
          identifiers: "Easun SMH-II-0F6C29"
`

ivan-rio, 2024-12-30:
I see that there is a trigger for update static parameters in source code in PI_Serial.cpp, but unfortunately sometimes it did not update to new values, so i added `requestStaticData = true;` in 120 line of PI_Serial.cpp to force the update of static values all time.

closed 2024-12-30 by ivan-rio