softwarecrash/Solar2MQTT-ESP8266 #200: [Bug]: Sometimes mqtt reports stucks for a while

Sasha2018RL, 2025-01-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Used Hardware?

Other (please write your hardware in 'What happened?')

### What happened?

Hello! I use this software with esp8266 12E NodeMCU, MAX232 as rs232-ttl converter and xl6009 dc-dc to power from inverter monitoring port. Inverter is PowMr 3.2Kw. Sometimes esp lost connection with inverter, led blinks 2 times. I fixed it using shielded cable and grounded its shield. Now in web panel and /livejson data appears normally, but in mqtt reports sometimes it appears every 15 seconds as set in config, sometimes it stuck from few minutes to few hours

### Screenshots / Fotos

[image]
Config page
[image]
Main page

### Version

1.x.x and above

### Relevant livejson output

```json
{"EspData":{"Device_name":"Solar","ESP_VCC":3.002,"Wifi_RSSI":-18,"sw_version":"2.0.0-Pre6A2","Free_Heap":11376,"HEAP_Fragmentation":21,"runtime":119019,"ws_clients":1,"detect_protocol":2,"detect_raw_qpi":"PI30"},"DeviceData":{"AC_In_Rating_Voltage":230,"AC_In_Rating_Current":13,"AC_Out_Rating_Voltage":230,"AC_Out_Rating_Frequency":50,"AC_Out_Rating_Current":13,"AC_Out_Rating_Apparent_Power":3000,"AC_Out_Rating_Active_Power":3000,"Battery_Rating_Voltage":24,"Battery_Recharge_Voltage":25.5,"Battery_Under_Voltage":24,"Battery_Bulk_Voltage":28.3,"Battery_Float_Voltage":27.7,"Battery_Type":"User","Current_Max_AC_Charging_Current":2,"Current_Max_Charging_Current":30,"Input_Voltage_Range":"Appliance","Output_Source_Priority":"SBU first","Charger_Source_Priority":"Only solar charging permitted","Parallel_Max_Num":1,"Machine_Type":"Off Grid","Topology":"Transformerless","Output_mode":0,"Battery_Redischarge_Voltage":27,"PV_OK_Condition_For_Parallel":0,"PV_Power_Balance":1,"Device_Model":"VMII-NXPW5KW","Protocol_ID":"PI30","Buzzer_Enabled":true,"Overload_Bypass_Enabled":false,"Power_Saving_Enabled":false,"LCD_Reset_To_Default_Enabled":false,"Overload_Restart_Enabled":false,"Over_Temperature_Restart_Enabled":false,"LCD_Backlight_Enabled":true,"Primary_Source_Interrupt_Alarm_Enabled":false,"Record_Fault_Code_Enabled":true},"LiveData":{"AC_In_Voltage":222.1,"AC_In_Frequenz":49.9,"AC_Out_Voltage":230,"AC_Out_Frequenz":50,"AC_Out_VA":162,"AC_Out_Watt":85,"AC_Out_Percent":5,"Inverter_Bus_Voltage":381,"Battery_Voltage":25.66,"Battery_Charge_Current":0,"Battery_Percent":77,"Inverter_Bus_Temperature":37,"PV_Input_Current":0.1,"PV_Input_Voltage":25.7,"Battery_SCC_Volt":0,"Battery_Discharge_Current":3,"Status_Flag":10000,"Battery_Voltage_Offset_Fans_On":0,"EEPROM_Version":0,"PV_Charging_Power":0,"Device_Status":10,"Battery_Load":-3,"PV_Input_Power":0,"Inverter_Operation_Mode":"Battery"}}
```

### What browsers are you seeing the problem on?

no Issue with the Browser or WebUI

softwarecrash, 2025-01-06:
Please try the previus Release Version before the iOS bugfix

Sasha2018RL, 2025-01-06:
On last release version i had some issues with connection to inverter, so have updated to 2.0.0Pre-6A2

softwarecrash, 2025-01-07:
[Solar2MQTT_d1_mini_2.0.0-Pre6A5.bin.zip](https://github.com/user-attachments/files/18330027/Solar2MQTT_d1_mini_2.0.0-Pre6A5.bin.zip)
please try this

Sasha2018RL, 2025-01-07:
Thank you, mqtt posting works as set in config. 
But there is some error in response
solar/RAW/QPIWS: Wrong Length(31), Contact Dev:0000010000000000000000000000000

softwarecrash, 2025-01-07:
what for a device you have?

softwarecrash, 2025-01-07:
[Solar2MQTT_esp01_1m_2.0.0-Pre6A6.bin.zip](https://github.com/user-attachments/files/18332414/Solar2MQTT_esp01_1m_2.0.0-Pre6A6.bin.zip)
i think this will fix it