softwarecrash/Solar2MQTT-ESP8266 #201: [FEATURE] Send control commands using REST API

Sasha2018RL, 2025-01-06:
### Is your feature request related to a problem? Please describe

In some cases sending commands to inverter via mqtt is not easy to use. For example, I use /livejson to get data from inverter and dont want to use MQTT at all 

### Describe the solution you'd like

I think will be good to add some request like /execute?command=<command here>, for example 
POST <ip>/execcute?command=POP02
Response: ACK

### Describe alternatives you've considered

Would be good to send command in one http request, wait for invertor's answer and get it in a response

softwarecrash, 2025-01-07:
Hello,
that is allready implemented, but not follow the REST API or other logic.
badly i forgot to document it in the wiki, will write it now
`http://<DEVICE_IP>/set?CC=<yourCommand>`

Sasha2018RL, 2025-01-07:
Thank you!
 I've tried it and got answer "Message received", but how can I get inverter’s answer to command? Probably there is some other request, that waits command to complete?

softwarecrash, 2025-01-07:
the "Message received" is the answer that it have collected in the worker queue, so unfurtanly its not possible to get the device andwer direct over that webhook, why? xour command comes in the queue and will send to the device but when a bunch other commands wait the webserver will wait for this answer and that is too long to answer your request and the webserver will crash (watchdog timeout)
i can try to implement the answer in it, but i dont have a working device actual, so i can only send you a test

Sasha2018RL, 2025-01-07:
Okay, I can test it. Probably it could be fixed by increasing watchdog timeout to 8 seconds for example? Device got request, puts command to the queue and when its completed - sends response to http client

softwarecrash, 2025-01-07:
just checked, i cant get the device answer in the webserver request section, so you need another way to check it by request the data or json again and check in static data section if the data you modified is now there.

modify the watchdog is a realy bad idea, the watchdog acts in microseconds and will cause a error in some sections when a function need more than like 500ms
the code works with async systems, so nothing can wait for other thingsy when incerase the wait time to that high amount you will get a uncontrolable behavior and a device zombi until it crashes

softwarecrash, 2025-01-07:
that need a asyncwebclient that is not implemented actual and i think its too much for this function

Sasha2018RL, 2025-01-10:
Thank you for help!

closed 2025-01-10 by Sasha2018RL